256dpi/arduino-mqtt #135: Board Library for esp8266 2.5.0-beta2 does not work with WiFiClientSecure

Urs-Eppenberger, 2018-12-20:
Hello Joel, just to let you know:
I recently upgraded the esp8266 board library to 2.5.0-beta2
Your library works for normal connections, but not for TLS using WiFiClientSecure, if the above mentioned board library is installed.

In the log of my mosquitto broker (on Raspberry Pi) I see, that the device is trying to connect:
`New connection from 192.168.11.84 on port 8883`
On the device  side, client.connect() never becomes true.
I tried to ignore this test and submitted a payload to a topic on the broker, it never arrived there.
It seems safe to assume, that the connection somehow is not correctly working

I downgraded to 2.4.2, everything is back to normal and working flawlessly.

From what I read on the github page of the ESP8266 project, they did a lot of work under the hood. Probably it is best to wait a few days or even weeks before digging into this, but this is up to you.

luisgcu, 2019-01-12:
> Hello Joel, just to let you know:
> I recently upgraded the esp8266 board library to 2.5.0-beta2
> Your library works for normal connections, but not for TLS using WiFiClientSecure, if the above [...]

Hi, I have similar issue with the ESP32, I can't connect to Mosquitto  SSL using this library which in my opinion is very well done. 
Any way to solve that?

debsahu, 2019-01-17:
Please check this: https://github.com/debsahu/ESP_MQTT_Secure hope it helps!

Teddyz, 2019-02-17:
I can confirm that 2.5.0 works with neither arduino-mqtt nor pushover (that I use and recommend for sending messages from sensors to my phone). 2.4.2 works.

BSGEngineering, 2019-02-28:
Working on this now - I think I'm close to a solution

debsahu, 2019-02-28:
I use this library (and PubSubClient) to connect to AWS-IoT, CA certificate + private certificate + private key https://github.com/debsahu/ESP-MQTT-AWS-IoT-Core Hope this helps, but connecting to mosquitto is the same idea.

BSGEngineering, 2019-02-28:
An example that might work for you... I pulled from the BearSSL Validation example here: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino

I use the fetchFingerprint(); section to create a secure connection, leave it open and return to connect to MQTT using this library secure example. Alternatively, I believe you could pull any of the other certification methods from the validation example and replace that section of code with the desired certification check. 

Variables that need updating:
const char ssid[] = ""; //WiFi SSID
const char pass[] = ""; //WiFi Password
const char clientId[] = ""; //MQTT ClientID
const char username[] = ""; //MQTT Username (if any)
const char password[] = ""; //MQTT Password (if any)

static const char fp[] PROGMEM = ""; //SHA1 Fingerprint
const char *   host = ""; //MQTT Host address

client.subscribe("/hello", 2); //Subscribe Topic, QOS

Let me know if this works/if you have questions!

[Secure_MQTT_BearSSL.txt](https://github.com/256dpi/arduino-mqtt/files/2916673/Secure_MQTT_BearSSL.txt)

BSGEngineering, 2019-03-02:
I updated the example to include the two other certificate verification methods from the BearSSL example - just update the cert info and comment in/out the desired method. I have no way to check the functionality for those types of certs at the moment , but it should work!

256dpi, 2019-03-12:
I just updated the example and added `net.setInsecure()` with a note pointing to the available validation methods. I'm happy to accept PR's that add more options (commented out). Thanks for the research everyone!

closed 2019-03-12 by 256dpi