256dpi/arduino-mqtt #138: client publishes LWT at wrong time

abcd-ca, 2019-01-05:
I have `client.setWill(topic_connected, "false", true, LWMQTT_QOS1);` before connect. 

Once connected, I send `client.publish(topic_connected, "true", true, LWMQTT_QOS1);`

In this way, I can have a dashboard that subscribes to `topic_connected` so that I can glean whether or not my device is connected (note bool payload difference). 

When I unplug my device, the broker does publish the LWT as expected. The problem is, right after my `client.publish`, the LWT is also published in that case – LWT in quick succession with "false" payload after the intended publish with "true" payload. There is no indication that my client has actually disconnected so I don't know why the message gets triggered.

Maybe I'm missing something I did wrong?

abcd-ca, 2019-01-05:
I was calling `setWill` before `connect` because [in the readme](https://github.com/256dpi/arduino-mqtt/blob/47ef2bb128df2188162877fe01f2d363b5490335/README.md#api) it advises to do so. I tried moving it to after the connect and the LWT defect seems to be gone. Are the docs out of date? I can make the tweak there if that's easiest. Otherwise I'd appreciate assurance that my code really is the way it should be now, in case there's some problem I'm unaware of with my fix.

256dpi, 2019-01-22:
Can you post the complete code?

abcd-ca, 2019-01-22:
Sure @256dpi, thanks for taking a look

```C++
#include <EEPROM.h>
#include <ESP8266WiFi.h>
#include <MQTT.h>
#include "credentials.h"

// TODO clean up: separate into multiple files

const char ssid[] = WIFI_SSD;
const char pass[] = WIFI_PASS;

const char mqtt_broker_user[] = BROKER_USER;
const char mqtt_broker_password[] = BROKER_PASS;

uint8_t deviceId = 1;

char broker_client_id[8];

MQTTClient client;

unsigned long lastMillis = 0;

const uint8_t topicBuffLimit = 50;

// see initStartupTopics
// char topic_config[topicBuffLimit];
char topic_connected[topicBuffLimit];

/**
 * Topics whose string is comprised of variables determined at runtime
 */
void initStartupTopics() {
  Serial.println(F("initStartupTopics"));
  snprintf(topic_connected, topicBuffLimit, "scnr/%03d/connected", deviceId);
  Serial.print(F("  topic_connected: ")); Serial.println(topic_connected);
}

void connect() {
  Serial.print("\nConnecting to broker...");
  while (!client.connect(broker_client_id, mqtt_broker_user, mqtt_broker_password)) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println(F("\nbroker connected!"));
  	
  client.setOptions(10, false, 1000);
}

void setup() {
  Serial.begin(115200);

  // note wifiClient stripped from example for simplicity
  client.begin("192.168.0.102", wifiClient);

  initStartupTopics();

  connect();
  
  client.setWill(topic_connected, "false", true, LWMQTT_QOS1);
  
  // publish connect msg ('last will' will publish this as false)
  client.publish(topic_connected, "true", true, LWMQTT_QOS1);
}

void loop() {
  client.loop();
  delay(100);  // <- fixes some issues with WiFi stability

  if (!client.connected()) {
    connect();
  }
}
```

abcd-ca, 2019-06-05:
I have since simplified my code and I'm not sure what about it is different but LWT seems to work fine now.

Pseudo-code for anyone else' benefit, here's the order I do things in:

*in `setup`:*

1. `client.begin`
2. `client.setOptions`
3. `client.setWill`
4. `client.onMessage`
5. connect as in [this example](https://github.com/256dpi/arduino-mqtt/tree/master/examples/AdafruitHuzzahESP8266)
6. `client.subscribe`

*in `loop`:*

1. `client.loop`
2. reconnect if `!client.connected()` (call connect again as in [this example](https://github.com/256dpi/arduino-mqtt/tree/master/examples/AdafruitHuzzahESP8266))

closed 2019-06-05 by abcd-ca