256dpi/arduino-mqtt #140: Add ArduBadge badge for new users

gilmaimon, 2019-01-20:
Hi!

A nice addition to the readme page can be a badge from [ArduBadge](https://www.ardu-badge.com/), It will look like this:
[image]
Markdown:
`[[image]](https://www.ardu-badge.com/MQTT)`

The badge shows:
1. That MQTT is available from the Arduino IDE (Library Manager)
2. The newest available version
3. Clicking on the badge redirects users to a custom page with explanation on installing MQTT (from the ide and from a ZIP file). It is great for new users.

Also, badges are awesome.
Thank You! :)

256dpi, 2019-02-08:
I very much appreciate your work and the idea behind your project. However, I'm personally not a big fan of all these vanity badges and only add the ones related to the code (CI build, release) to my projects. I hope you can understand that.

closed 2019-02-08 by 256dpi

gilmaimon, 2019-02-08:
> I very much appreciate your work and the idea behind your project. However, I'm personally not a big fan of all these vanity badges and only add the ones related to the code (CI build, release) to my projects. I hope you can understand that.

Of course I do, and I really appreciate your kind reply and the work you do for OS in general and specifically Arduino.
Thank you :)