256dpi/arduino-mqtt #144: Maximun keep alive time

jmpmscorp, 2019-01-30:
Hi again @256dpi .

I continue testing my project and I have noticed there is a "problem" when you set keep alive higher than 86 seconds. When I set keep alive with setOptions at 90 seconds, pingreq/pingresp cicle is done around 4 seconds delay. If I set it to 120 seconds, cicle is done around 25 seconds. I have been testing with UNO + Ethernet Shield with default millis() as clock source. 

What could be the reason?

256dpi, 2019-02-08:
I released a new version that hopefully fixed the bug. Would you be able to test it?

jmpmscorp, 2019-02-08:
I have been testing new release with UNO + ETHERNET SHIELD with two different values, 150 and 300 seconds. I can confirm that this bug is resolved.

Thanks again @256dpi.

256dpi, 2019-02-09:
Cool thanks!

closed 2019-02-09 by 256dpi