256dpi/arduino-mqtt #145: Chunked Reading and Writing

256dpi, 2019-02-08:
There have been various requests for adding a chunked reading and writing API to process messages that are larger than the configured buffer size. This issue is to track the implementation of that.

mih380, 2019-04-27:
Is there a beta library available to test the chunked message functionality?

256dpi, 2019-04-28:
Not yet, the functionality has not yet been implemented in the underlying lwmqtt library.

nickels, 2020-03-21:
> Not yet, the functionality has not yet been implemented in the underlying lwmqtt library.

hi, using the library. It's great, thanks. I see you're also maintaining the underlaying library. When do you forsee that chunking will be implemented?

nickels, 2020-03-21:
> > Not yet, the functionality has not yet been implemented in the underlying lwmqtt library.
> 
> hi, using the library. It's great, thanks. I see you're also maintaining the underlaying library. When do you forsee that chunking will be implemented?

Was able to fix my issue by increating the buffer like so:

`MQTTClient MqttClient = MQTTClient(1024);`