256dpi/arduino-mqtt #147: Support class member as an onMessage callback

corporateanon, 2019-02-11:
I have a class which uses `MQTTClient`. I want to pass a pointer to class method as a callback to `onMessage`. Could you please create an overloaded `onMessage` method which would accept a "context" (a pointer to a class instance) and a callback (a pointer to a method of this class)?

Otherwise it is almost impossible for me to have a callback which would have an access to the rest of my class instance methods.

```cpp
class App
{
  //...skipped
  public:
    void messageHandler(String &topic, String &payload);
    void setup();
    MQTTClient client;
}

//---

void App::setup()
{
  //...skipped
  client.onMessage(this, App::onMessage);
}
```

256dpi, 2019-02-12:
I'm resistant to changing the callback signatures again and breaking already existing code.

Would a `void * ref` property on the `MQTTClient` class suffice? Together with the advanced callback you could get a reference to your custom class:

```c++
void messageReceived(MQTTClient *client, char topic[], char payload[], int payload_length) {
  auto parent = (*App)client.ref;
}
```

corporateanon, 2019-02-12:
Yes, it would be very nice!

abcd-ca, 2019-05-28:
I would also benefit from a ref on the simple callback signature

chris-zen, 2019-08-18:
@256dpi one solution that wouldn't break backward compatibility is to add a new overloaded `onMessage` that has a `lambda` as a parameter. Lambdas allow to capture other variables from the environment, for example `this` which would solve the problem with class members. The interface would be something like:
```c++
#include <functional>

void onMessage(std::function<void(String&, String&)> callback) {}
```

256dpi, 2019-12-02:
Added the `ref` property in d0c2820.

closed 2019-12-02 by 256dpi

ramboerik, 2020-01-17:
Thanks for implementing this, found it very useful!

jenschr, 2023-10-31:
Thanks for making this and adding this feature. It wasn't instantly obvious to me how to use this, so I made a bare-bones example showing how to implement a wrapper class using MQTTClientCallbackAdvancedFunction and ref  https://github.com/jenschr/BasicMqttClassExample Feel free to link to it in the docs.

256dpi, 2023-10-31:
@jenschr Thanks for letting me know! If you want, we can also add the example to the examples within this repository. You could check out the existing examples, and submit a PR for one with your changes.

jenschr, 2023-11-01:
Yeah. PR is the way to go. I can make it an Arduino project before pushing, so it's in line with the other examples.