256dpi/arduino-mqtt #154: SSL not working properly disconnects after few seconds and never connects back

ShivamJoker, 2019-03-07:
when i reboot my deivce the mqtt connects and after few second it gets disconnected and when it tried to reconnect it just throws the error

```
[E][ssl_client.cpp:33] handle_error(): UNKNOWN ERROR CODE (004C)
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -76
[E][ssl_client.cpp:33] handle_error(): UNKNOWN ERROR CODE (004C)
[E][ssl_client.cpp:35] handle_error(): MbedTLS message code: -76

```

Mumie-hub, 2019-03-11:
hi guys, could be related to Breaking Change in Arduino Release 2.5.0 [https://github.com/esp8266/Arduino/releases](url) "modern SSL/TLS with BearSSL (now default)"

ShivamJoker, 2019-03-12:
@Mumie-hub yes I am using platform io with updated framework and it's on esp32

a7md0, 2019-07-19:
I have the same problem using platformio and ESP32 Arduino.

Espressif32 => 1.9.0

BSGEngineering, 2019-07-20:
Related to mumie-hub's response, did u try using the updated secure example, or the updated example in my pull request?

ThierryVC, 2019-12-04:
I'm getting the same error 
`[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)`
when I call `WiFi.disconnect(true, true);`
It disappears if I call `disconnect()` beforehand on the mqtt client object.

Drazyl67, 2020-03-03:
Any update for this issue?
ESP32 with platformio

clausqr, 2020-07-17:
From https://github.com/ARMmbed/mbedtls/blob/mbedtls-2.4.2/include/mbedtls/net_sockets.h#L42
``` cpp
#define MBEDTLS_ERR_NET_RECV_FAILED                       -0x004C  /**< Reading information from the socket failed. */
```

Also have a look at https://github.com/curl/curl/issues/1327

256dpi, 2020-10-18:
Seems to have been fixed in ESP32. Please reopen if problem persists.

closed 2020-10-18 by 256dpi

lrodorigo, 2021-03-07:
Still getting these random issues with `espressif32@3.0.0`  and `256dpi/MQTT@^2.4.8`.

I am using this library from GoogleCloudIoTCore wrapper.

I get (randomly) both:

`[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-80) UNKNOWN ERROR CODE (0050)`

and

`[E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C)`

dforsyth1, 2021-03-20:
I'm getting this with knolleary's MQTT library (004C).  Haven't see the 0050.

Seems the idea of switching this doesn't make sense...

Mault7, 2021-09-30:
hola como están comunidad les comento que solucione ese problema de [E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C) se debe que el puerto esta en disputa hay dos eventos que quieren el apropiarse al puerto y cuando el WiFiClientSecure no lo obtiene manda el mensaje UNKNOWN ERROR CODE (004C) mismo que significa Error al leer la información del socket como manda el link del compañero "clausqr " para ello se pueden usar un semaphremutex para hacer exclusion mutua y proteger el recurso.
bueno mas nada espero y les sirva saludos a todos

raphael-bmec-co, 2022-03-18:
@lrodorigo I am seeing the same thing. Any luck?

emanavas, 2022-05-12:
> hola como están comunidad les comento que solucione ese problema de [E][ssl_client.cpp:33] _handle_error(): [data_to_read():270]: (-76) UNKNOWN ERROR CODE (004C) se debe que el puerto esta en [...]

Hola @Mault7 , estoy con ese mismo problema y no sé como aplicar la solucion que planteas....
Me puedes mostrar donde y como agrego el Semaforo? (se como usar el RTOS).
Gracias,

jojo2massol, 2022-07-21:
I'm still getting this error on ESP32-WROOM-32E.
This happens when I send my request to nginx:
Error in context while displaying each line of the request :
```
Connection: close
[  9747][E][ssl_client.cpp:37] _handle_error(): [data_to_read():356]: (-76) UNKNOWN ERROR CODE (004C)

Hello, world
```
What I should get (this was obtained using curl):
```
< HTTP/1.1 200 OK
< Server: nginx/1.18.0 
< Date: Thu, 21 Jul 2022 14:36:41 GMT
< Content-Type: text/plain
< Content-Length: 13
< Connection: close
< 
Hello, world
* Closing connection 0
* TLSv1.3 (OUT), TLS alert, close notify (256)
```

I have to add that I'm doing a POST request, and for now, the end of my data received on the server is incomplete, I have 2 unknown characters, and not the closing braces of my JSON. I don't think this is linked, but it's here.

NorthernMan54, 2022-11-16:
Solved - leaving this here for others who stumble upon this

I came across this issue as well on my project, and after investigating further what was happening I found that invoking a mqtt publish from a separate task was causing the issue. My code was using both core's and multiple tasks, and publishing from a separate task caused the issue.  To resolve / workaround the issue I queued the message, then used the main loop to dequeue the messages and publish them.

robikovacs, 2023-07-01:
Thank you @NorthernMan54! You saved hours of debugging for me.