256dpi/arduino-mqtt #158: arduino mkr1000 loses connection with mqtt  after few hours

cr0wl33, 2019-05-07:


cr0wl33, 2019-05-07:
#include <PubSubClient.h>
#include <WiFi101.h>

void callback(char* topic, byte* payload, unsigned int length);

//EDIT THESE LINES TO MATCH YOUR SETUP
#define MQTT_SERVER  "secret_ip"     //you MQTT IP Address
const char* ssid = "secret1";
const char* password = "secret2";

const int switchPin1 = 1;
const int switchPin2 = 2;
const int switchPin3 = 3;
const int switchPin4 = 4;
const int switchPin5 = 5;
const int switchPin6 = 6;
const int switchPin7 = 7;
const int switchPin8 = 8;

//These are the MQTT Topic that will be used to manage the state of Relays 1 ~ 8/
//feel free to replicate the line if you have more relay switch to control, but dont forget to increment the number suffix so as increase switch logics in loop()

char const* switchTopic1 = "/house/switch1/";
char const* switchTopic2 = "/house/switch2/";
char const* switchTopic3 = "/house/switch3/";
char const* switchTopic4 = "/house/switch4/";
char const* switchTopic5 = "/house/switch5/";
char const* switchTopic6 = "/house/switch6/";
char const* switchTopic7 = "/house/switch7/";
char const* switchTopic8 = "/house/switch8/";

WiFiClient wifiClient;
PubSubClient client(MQTT_SERVER, 1883, callback, wifiClient);

void setup() {
  //initialize the switch as an output and set to LOW (off)
  pinMode(switchPin1, OUTPUT); // Relay Switch 1
  digitalWrite(switchPin1, LOW);

  pinMode(switchPin2, OUTPUT); // Relay Switch 2
  digitalWrite(switchPin2, LOW);

  pinMode(switchPin3, OUTPUT); // Relay Switch 3
  digitalWrite(switchPin3, LOW);

  pinMode(switchPin4, OUTPUT); // Relay Switch 4
  digitalWrite(switchPin4, LOW);

  pinMode(switchPin5, OUTPUT); // Relay Switch 4
  digitalWrite(switchPin5, LOW);

  pinMode(switchPin6, OUTPUT); // Relay Switch 4
  digitalWrite(switchPin6, LOW);

  pinMode(switchPin7, OUTPUT); // Relay Switch 4
  digitalWrite(switchPin7, LOW);

  pinMode(switchPin8, OUTPUT); // Relay Switch 4
  digitalWrite(switchPin8, LOW);

  //start the serial line for debugging
  Serial.begin(115200);
  delay(100);

  //start wifi subsystem
  WiFi.begin(ssid, password);
  //attempt to connect to the WIFI network and then connect to the MQTT server
  reconnect();

  //wait a bit before starting the main loop
[... 192 lines cut ...]
        client.subscribe(switchTopic8);
        //Do not forget to replicate the above line if you will have more than the above number of relay switches
      }

      //otherwise print failed for debugging
      else{Serial.println("\tFailed."); abort();}
    }
  }
}

//generate unique name from MAC addr
String macToStr(const uint8_t* mac){

  String result;

  for (int i = 0; i < 6; ++i) {
    result += String(mac[i], 16);

    if (i < 5){
      result += ':';
    }
  }

  return result;
}

cr0wl33, 2019-05-07:
above is the code i use  it works only  after few hours it not working any more
i think it has something to do with connecting to the mqtt server ?

rokrodic, 2019-10-14:
#include <PubSubClient.h> ?

256dpi, 2019-10-14:
@cr0wl33 PubSubClient is another MQTT library for Arduino. You can find it here: https://github.com/knolleary/pubsubclient.

closed 2020-10-18 by 256dpi