256dpi/arduino-mqtt #162: mqtt tls LWMQTT_NETWORK_FAILED_WRITE

dmet6789, 2019-06-29:
Hello
I'm using yun rev2 arduino and try to connect with mosquitto broker running in a windows pc.
the mosquitto broker is configured to accepts user and password to connect with it and use 8883 port.

when i try to connect to the mosquitto server with this sentence
 while(!clienteMQTT.connect("dcdCliente",usuario,password,true)){

i receive error code -6 LWMQTT_NETWORK_FAILED_WRITE
and Return code = 6 

thanks in advance for your help
best regards
diego

dmet6789, 2019-06-29:
The complete sketch that i'm using:
#include <Bridge.h>
#include <BridgeSSLClient.h>
#include <MQTT.h>
#include <Process.h>

const char broker[] = "192.168.1.130";
const int port = 8883;
const char usuario[] = "dcduser";
const char password[] ="password";

int LDRPin = A0;

BridgeSSLClient clienteSSL;
MQTTClient clienteMQTT;

 
int V;
String mensaje;
char* val1;
char val2[] = "OTRO VALOR";

void conectaBroker(){
   Serial.println("Conectando...");
   while(!clienteMQTT.connect("dcdCliente",usuario,password,true)){
   //while(!clienteMQTT.connect("SoyYoDCD")){
    
    //Serial.println(lwmqtt_err_t);
    int lwmqtt_err_t = clienteMQTT.lastError();
    int lwmqtt_return_code_t = clienteMQTT.returnCode();
    Serial.println("err: ");
    Serial.print(lwmqtt_err_t);
    Serial.println();
    Serial.println("RC: ");
    Serial.print(lwmqtt_return_code_t);
    Serial.print(".");
    delay(1000);
   }
   Serial.println("Conectado !!!");
  
}
 
void setup() 
{

  //saber la ip
//  Process p;
//  p.runShellCommand("/sbin/ifconfig eth1 | grep 'inet addr:' | cut -d: -f2 | awk '{print $1}'");
//  Serial.println("IP eth1: ");
//  while (p.available()){
//    char c = p.read();
//    Serial.print(c);
//   }
//   Serial.flush();

   Bridge.begin();
   Serial.begin(9600);
   clienteMQTT.begin(broker, port, clienteSSL);
   conectaBroker();
   
}
 
void loop()
{

  clienteMQTT.loop();    //envía los paquetes
  if(!clienteMQTT.connected()){
    conectaBroker();
  }
  V = analogRead(LDRPin); 
//  mensaje = V;        
  Serial.println(V);
  itoa(V,val1 ,10);      //convertir int to string
  Serial.println(val1);
  mensaje.concat(val1);
  mensaje.concat(val2);
  Serial.println(mensaje);
  clienteMQTT.publish("/DcdMqtt",mensaje);
  delay(1000);
}
thanks 
regards
diego

256dpi, 2019-08-18:
Sorry for the late reply. You're are passing "true" to `connect` which results in skipping the network level connection initiation:

```
clienteMQTT.connect("dcdCliente",usuario,password,true) // <---
```

The skip parameter is for scenarios where you want to create a connection yourself. Setting `false` will allow the library to do this automatically.

closed 2019-08-18 by 256dpi