256dpi/arduino-mqtt #165: Blocking publish with esp8266/Arduino v2.5.x

Teddyz, 2019-08-10:
Short periods when loop() is not running while publishing topics is a problem for me because I at the same time tries to run a stepper motor. 

To show the problem, just run the example AdafruitHuzzahESP8266Secure with added:

In setup add:
pinMode(LED_BUILTIN, OUTPUT);
net.setInsecure();

In loop add:
digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));

The LED will flicker at every publish, when using esp8266/Arduino v2.5.x. Using previous version this is not a problem.  I am not suggesting that this is an error in arduino-mqtt, but would like to hear if there is a known way to mitigate this problem.

256dpi, 2019-08-18:
My first assumption would be that this is due to the synchronous nature of the Arduino APIs. Publishing a message is synchronous and will do all the work immediately and then return. If this takes a lot of time the code would block for the whole period.

That said, a simple Publish with QoS0 should still be very fast as this only requires to send the message bytes to the modem. However, already a publish with QoS1 will wait for the ACK from the server, which depending on the connection might take several milliseconds to receive.

Could you post the exact code you used? Also the "flickering" of an LED is not a very good performance indicator... If you could measure the time to publish using `millis()` we could see if there is really a difference between the mentioned versions.

256dpi, 2019-09-13:
If you're not using `net.setInsecure();` you need to use one of methods to verify the server certificate since the ESP82667 does not have the root CAs installed:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino

I just ran the unmodified ESP8266 examples at home on a 1Gpbs WPA2 network connecting to shiftr.io. The unsecured example has a latency up to 1ms and the secured up to 60ms. I guess this is normal for any micro-controller doing unaccelerated TLS cryptography.

This timing is also heavily dependent on the cryptographic parameters of the server TLS certificate.  A 512bit RSA key needs more computation than a 256bit key or the EC 512/256 counterparts.

Teddyz, 2019-10-07:
Thanks for the explanation! But it gives me so many follow up q:s... I hope you can answer a few.
1) You got the code blocked 60ms when sending QoS=0 using ESP8266 @80 MHz? I suppose you use esp8266/Arduino v2.5.2. My result was 170ms... Strange. 

2) I use `net.setInsecure();` Okay, that means vulnerable to MITM attacks, but I suppose it is still encrypted? 

3) My certificate is from Lets Encrypt: Using my web browser it looks like this:
[image]
I assume 256 bits.  Then it should be similar timing. Or can it be the old Pentium 4, 2.8GHz computer that runs Mosquitto on Win XP that causes the difference?

4) This is really outside your scope, but do you have any idea why the delay was negligible  when I ran Arduino 2.4.2?

5) Do you have a link to an example or pseudo-code on how to handle  the secure connection separate from arduino-mqtt?

Thanks!

Teddyz, 2020-02-07:
The problem is the same for me in Arduino v2.6.3.  It now takes ~200ms to publish a single QoS 0 message.

How can I use Serial.print in client.c? It works in MQTTClient.cpp, but all my tests to mimic the header-includes to include Serial fails.  Or is there a better way to get understanding on where it fails?

Changing ESP8266 from 80 to 160 MHz and/or using "v2 Higher Bandwidth" makes no difference.

Most of the actual delay seems to be on the computer running mosquitto.exe. This is how it looks when I run v2.6.3:
[image]

And this is how it looks when I run v.2.4.2:
[image]
The delay on the mosquitto-side seems equal (to my untrained eye), but the blocking is very short on the ESP when running Arduino v2.4.2. 

My question is if there is anything that can be done in this library to mitigate the blocking?

256dpi, 2020-02-08:
No there is nothing we can do, the underlying client should be fixed.

Teddyz, 2020-02-08:
OK. I found a workaround in changing so the Windows computer sends ACK immediately. It improves the situation vastly at the unknown cost of other TCP communication going slow on that computer.

On Win XP (hopefully just me still using it) or Win7, add this to the registry:
Subkey: HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\Tcpip\Parameters\Interfaces\<Interface GUID>
Entry: TcpAckFrequency
Value Type: REG_DWORD, number
Valid Range: 0-255
Default: 2 

Set the value to 1. Add it to Favourites so you easily can revert if this is fixed in Arduino later. Then reboot for it to take effect.
[https://support.microsoft.com/en-us/help/328890/new-registry-entry-for-controlling-the-tcp-acknowledgment-ack-behavior](https://support.microsoft.com/en-us/help/328890/new-registry-entry-for-controlling-the-tcp-acknowledgment-ack-behavior)

closed 2020-02-08 by Teddyz