256dpi/arduino-mqtt #166: LWT status turns into offline then never gets back to online

peros550, 2019-08-19:
I am using this nice library and I need some help. Everything works great except the LWT part. 

The problem is that after a while, my broker (mosquitto on a raspberry) shows the Arduino board as **offline** for no good reason.  During that time the Arduino is actually running fine  ,sending and receiving mqtt messages that control relays as expected. 

I suspect that there is no part in the code (except in the very beginning setup() function) that tells the broker that Arduino is back online. 

Do you think the following command should be added to connect() function?

`client.publish("tele/ArduinoMega1/LWT", "Online");`

Here is my simplified code related to mqtt: 
```
#include <Ethernet.h>
#include <avr/wdt.h>
#include <MQTT.h>

//Ethernet Variables
byte mac[] = {0x90, 0xA2, 0xDA, 0x0D, 0x44, 0x6F};
IPAddress ip(192, 168, 177, 27);
IPAddress myDns(192, 168, 177, 1);
IPAddress gateway(192, 168, 177, 1);
IPAddress subnet(255, 255, 255, 0);
//*****

EthernetClient ethClient;
MQTTClient client;

const char * mqttClientId = "Arduino-Client";
const char * mqttUsername = "username";
const char * mqttPassword = "password";
const char* mqtt_server = "raspberry";
long mqtt_PreviousMillis = 0;
long mqtt_PreviousMillis1 = 0;
long mqtt_update_interval = 5000; //
long lastMQTTReconnectAttempt = 0;
 

void connect() {
Serial.print("connecting...");
while (!client.connect(mqttClientId, mqttUsername, mqttPassword)) {
Serial.print(".");
delay(1000);
   
  }

  Serial.println("\nconnected!");
  client.subscribe("cmnd/output1/POWER");
     
}

void messageReceived(String &topic, String &payload) {
  Serial.println("incoming: " + topic + " - " + payload);

//Do something with received message
}

void setup() {

//Start Serial
Serial.begin(115200);
while (!Serial) {
 // wait for serial port to connect. Needed for native USB port only
}
Serial.print("start ....");

//Start Ethernet
Ethernet.begin(mac,ip, myDns, gateway, subnet);
delay(1500);
lastMQTTReconnectAttempt = 0;
Serial.print("server is at ''");
Serial.println(Ethernet.localIP());

//Start MQTT
client.begin(mqtt_server, ethClient);
client.onMessage(messageReceived);
client.setWill("tele/ArduinoMega1/LWT", "Offline", true, 2);
connect();
client.publish("tele/ArduinoMega1/LWT", "Online");

}

void loop() {
// put your main code here, to run repeatedly:
 client.loop();

   if (!client.connected()) {
  connect();
  }

  //Once every minute publish  status to the broker
  if(client.connected()){
    if  (  (millis() - mqtt_PreviousMillis1) > 60000){
    mqtt_PreviousMillis1 = millis();
    
    //Publish some mqtt messages 
    }
  }

}
```

256dpi, 2019-08-20:
If you run mosquitto in "verbose mode" it should tell you why the connection has been lost and the LWT as a result of it dispatched.

Also note, it seems that the LWT message "Offline" is retained, while the "Online" message is not. This will make the broker to only remember the "Offline" message and not the "Online".

peros550, 2019-08-20:
> If you run mosquitto in "verbose mode" it should tell you why the connection has been lost and the LWT as a result of it dispatched.
>
> Also note, it seems that the LWT message "Offline" is retained, [...]

Thank you for your quick response. I will try publishing the "Online" message in retained mode and let you know. 

I will report back as soon as I have news and update status of the issue.

peros550, 2019-08-20:
I do think that the following command should be part of the connect function: 

`client.publish("tele/ArduinoMega1/LWT", "Online",true);`

Here is  the proposed connect() function

```
void connect() {
Serial.print("connecting...");
while (!client.connect(mqttClientId, mqttUsername, mqttPassword)) {
Serial.print(".");
delay(1000);
}
client.publish("tele/ArduinoMega1/LWT", "Online",true);
Serial.println("\nconnected!");
//Subscribe to other topics as per the design logic
client.subscribe("cmnd/output1/POWER");
}
```

peros550, 2019-08-20:
Problem solved. Here is what worked for me: 

```
void connect() {
Serial.print("connecting...");
while (!client.connect(mqttClientId, mqttUsername, mqttPassword)) {
Serial.print(".");
delay(1000);
}
client.publish("tele/ArduinoMega1/LWT", "Online",true,2);
Serial.println("\nconnected!");
//Subscribe to other topics as per the design logic
client.subscribe("cmnd/output1/POWER");
}
```

closed 2019-08-20 by peros550