256dpi/arduino-mqtt #167: Value of `retain` is not actually retained on esp8266 (NodeMCU).

rajdeeprath, 2019-09-10:
I tried different methods as defined in the header file. But whenever I provide retain as true, it appears as `true` only once in Mosquitto broker. After that, it becomes `false` automatically. You can test this by using a simplistic code:` client.publish("dt/test", payload, true, 0) ;` and subscribing to same topic with a client such as MQTTBox.

256dpi, 2019-09-10:
This is the expected behaviour. The broker will always retain only the last received message and only forward that with retain=true when subscribing to the topic. Messages received directly via a subscription are sent with retain=false to indicate normal delivery.

rajdeeprath, 2019-09-10:
Thank you for the clarification.!! I thank you for your spectacular work.

closed 2019-09-10 by rajdeeprath