256dpi/arduino-mqtt #170: lastError() and returnCode() always return 0

viperk1, 2019-10-06:
Hey there, I'm trying to do some error logging for when the mqtt connection fails using lastError() and returnCode() but they're always returning 0. Am I using the functions incorrectly or is there something deeper here?

```
if (client.connect(host, mqtt_user, mqtt_pass)) {
      Serial.println("connected");
      return true;
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.returnCode());
      Serial.print(" last error=");
      Serial.print(client.lastError());
      Serial.printf(". Try MQTT again in %i ms\n",reconnectDelay);
      check_mq_interval = millis() + reconnectDelay;
      return false;
    }
```

Output message is always :
"Attempting MQTT connection...failed, rc=0 last error=0. Try MQTT again in 30000 ms"

The happy path works fine, but when I stop the mqtt server to test error handling I get the above.

Teddyz, 2019-10-07:
I have no answer, but I can show you my approach. I have the errors as arrays global:
```cpp
const char lwmqtt_err_text[][28] = { "SUCCESS", "BUFFER_TOO_SHORT", "VARNUM_OVERFLOW", "NETWORK_FAILED_CONNECT", "NETWORK_TIMEOUT", "NETWORK_FAILED_READ", "NETWORK_FAILED_WRITE", "REMAINING_LENGTH_OVERFLOW", "REMAINING_LENGTH_MISMATCH", "MISSING_OR_WRONG_PACKET", "CONNECTION_DENIED", "FAILED_SUBSCRIPTION", "SUBACK_ARRAY_OVERFLOW", "PONG_TIMEOUT", "UNKNOWN" };
const char lwmqtt_return_code_text[][25] = { "CONNECTION_ACCEPTED", "UNACCEPTABLE_PROTOCOL", "IDENTIFIER_REJECTED", "SERVER_UNAVAILABLE", "BAD_USERNAME_OR_PASSWORD", "NOT_AUTHORIZED", "UNKNOWN_RETURN_CODE", "UNKNOWN" };
```
Then I have this code called from the loop:
```cpp
int8_t logMQTTerrors() {
  static lwmqtt_err_t lastMqErr = LWMQTT_SUCCESS;
  static lwmqtt_return_code_t lastMqRet = LWMQTT_CONNECTION_ACCEPTED;
  lwmqtt_err_t mqErr = mq.lastError();
  lwmqtt_return_code_t mqRet = mq.returnCode();
  if ( lastMqErr != mqErr ) {
    log(2, p("LWMQTT error %d->%d(LWMQTT_%s)", lastMqErr, mqErr, lwmqtt_err_text[(-1)*mqErr]));
    lastMqErr = mqErr;
  }
  if ( lastMqRet != mqRet ) {
    log(2, p("LWMQTT returnCode %d->%d(LWMQTT_%s)", lastMqRet, mqRet, lwmqtt_return_code_text[mqRet]));
    lastMqRet = mqRet;
  }
}
```
Replace the log-commands with something like.
I never see any return-codes, but some errors are logged.
/Dick

256dpi, 2019-12-02:
A failed connect should now be properly reflected by `lastError`.

closed 2019-12-02 by 256dpi