256dpi/arduino-mqtt #171: SIM808 Example

Johanb-55, 2019-10-10:
It is possible an example using Adafruit FONA 808? Or SIM808? I already have a MKR1400 working with this library but i need tho change the modem by SIM808.

Johanb-55, 2019-10-13:
It is possible more info or an example? I have no experience with a wrapper class.

Johanb-55, 2019-10-23:
This is my code using simcom808

I connect to the server using "fona.TCPconnect("15.188.30.xxx",1883);" instead of "client.begin("15.188.30.xxx", net); it es possible?
I think my problem is in the "client.connect("d573b030-e5b9-11e9-b42a-79b8b21b1xxx", "aSzVhWxy3W9Bidkf8XXX", "")" with the token and id of the device. 

It is correct? what is the correct way to connect with this parameters?

Using this instruccions with  GSMclient and mkrgsm it works fine.

Thanks

`#include <Adafruit_SleepyDog.h>
#include <SoftwareSerial.h>
#include "Adafruit_FONA.h"
#include <MQTT.h>
 
#define FONA_RX 2
#define FONA_TX 3
#define FONA_RST 4

// this is a large buffer for replies
char replybuffer[255];

#include <SoftwareSerial.h>
SoftwareSerial fonaSS = SoftwareSerial(FONA_TX, FONA_RX);
SoftwareSerial *fonaSerial = &fonaSS;

Adafruit_FONA fona = Adafruit_FONA(FONA_RST);

uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);

uint8_t type;

//Adafruit_FONA net;
MQTTClient client;

void setup (){
  Serial.begin(115200);
  Serial.println(F("FONA basic test"));
  Serial.println(F("Initializing....(May take 3 seconds)"));

  fonaSerial->begin(4800);

  if (! fona.begin(*fonaSerial)) {
      Serial.println(F("Couldn't find FONA"));
      while (1);
    }
    type = fona.type();
    Serial.println(F("FONA is OK"));
    Serial.print(F("Found "));
    switch (type) {
      case FONA800L:
        Serial.println(F("FONA 800L")); break;
      case FONA800H:
        Serial.println(F("FONA 800H")); break;
      case FONA808_V1:
        Serial.println(F("FONA 808 (v1)")); break;
      case FONA808_V2:
        Serial.println(F("FONA 808 (v2)")); break;
      case FONA3G_A:
        Serial.println(F("FONA 3G (American)")); break;
      case FONA3G_E:
        Serial.println(F("FONA 3G (European)")); break;
      default: 
        Serial.println(F("???")); break;
    }
    
    // Print module IMEI number.
    char imei[16] = {0}; // MUST use a 16 character buffer for IMEI!
    uint8_t imeiLen = fona.getIMEI(imei);
    if (imeiLen > 0) {
      Serial.print("Module IMEI: "); Serial.println(imei);
    }

    Serial.println("ACTIVANT GPRS...");
    delay(100);
    fona.setGPRSNetworkSettings(F("TM"), F(""), F(""));
    delay(10000);
    
    // turn GPRS on
        if (!fona.enableGPRS(true)){
          Serial.println(F("Failed to turn on"));
        //break;
      }
  Serial.println("FET");
  delay(10000);

  
}

void loop(){
  Serial.print("ESTAT CONNEXIÓ...");
  Serial.println(fona.TCPconnected());
  if (!fona.TCPconnected()) {
    Serial.println("RECONECTAR...");
    Connectar();
  }
  else
  {
    Serial.println("CONNECTAT..."); 
  }
  Serial.print("VOLTA");
  delay(2000);  
}

void Connectar(){
  
  Serial.print("\nconnecting...");
  Serial.println("Connectar a thingsboard...");
  fona.TCPconnect("15.188.30.xxx",1883);
  
  //d573b030-e5b9-11e9-b42a-79b8b21b15c1
  //aSzVhWxy3W9Bidkf8XCX
  Serial.print(".......");
  delay(2000);

  
  while (!client.connect("d573b030-e5b9-11e9-b42a-79b8b21b1xxx", "aSzVhWxy3W9Bidkf8XXX", "")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");
  
  
}`

256dpi, 2023-02-04:
Closing as stale, please reopen if issue persists.

closed 2023-02-04 by 256dpi