256dpi/arduino-mqtt #178: QoS 1 logging (msg received, accepted or re-transmitted...)

salvq, 2019-11-24:
Hello, I switch to this mqqt library as I was looking for more visibility on QoS level especially to get visibility on whether the message was received by broker or was re-transmitted and so on.

I have enabled QoS level to 1 and I am using MKR GSM 1400 and so far I am not seeing any type of logging. That would be very help full for my local / arduino logging to SD card during testing phase or for whatever reason you migh use in sketch debugging.

This is what I used as QoS 1 testing setting:

```
  String payload;

  payload += "hello";
  payload += " ";
  payload += millis();
  payload += " ";
  payload += count++;
  payload += " ";
  payload += getCarrierSignal();
  client.publish("devices/testdevice/messages/events/", payload, false, 1);

```
Is there any callback or return about such a statuses ? Can you help me to enable whether message has been delivered, accepted or possible re-transmitted ?

Thank you

256dpi, 2019-11-25:
The `client.publish()` will perform the complete QoS1/2 flow and the returned value will indicate if the operation was successful. To analyze an error you can use `client.lastError()`.

salvq, 2019-11-25:
Got it, `lastError()` and `returnCode()` are printing 0 which is OK I guess.

1. How do I know that message has been delivered / accepted by Broker, if `LWMQTT_SUCCESS = 0` means packed is delivered to broker ?
2. When there is a re-transmission of message, how to read such a detail ?

256dpi, 2019-11-25:
1. You don't know. That is the idea of QoS0 (fire and forget).
2. There is no re-transmission, as there is no persistent session (yet).

salvq, 2019-11-25:
I probably did not stated my questions above correctly, my QoS level is set to 1 for message publish.

Given that, how do I know about successful delivery (at least one) and if transmission takes place ?

256dpi, 2019-11-25:
Please read the MQTT spec, it explains all these things very well.

salvq, 2019-11-25:
I did and I am aware what is level 0, 1 and 2 and how does it work.

Just would like to double check whether both parties (client and broker) **accepted** QoS 1 or 2 and following this.

What value in my sketch to check to see that broker sending back PUBACK packets ? Just to validate that I my message will get there at least once 1.

256dpi, 2019-11-25:
As I said the client will perform the full QoS1/2 flows. The details of the processing that happens in the background can be found here https://github.com/256dpi/lwmqtt/blob/master/src/client.c#L191.

salvq, 2019-12-04:
Got it, thanks

closed 2019-12-04 by salvq