256dpi/arduino-mqtt #181: Every 10th message, LWMQTT_BUFFER_TOO_SHORT

salvq, 2019-12-04:
Using latest version 2.4.4 and with every 10th message I am getting error -1 (LWMQTT_BUFFER_TOO_SHORT = -1,)

Can you help to identify an issue ?

**MQTT setup:**
```
BearSSLClient sslClient(gsmClient); // Used for SSL/TLS connection, integrates with ECC508
MQTTClient client;

client.begin("XXX.azure-devices.net", 8883, sslClient);

In loop:
client.loop();

Every 10s in loop:
Serial.print("before msg sent, lastError: ");
Serial.println(client.lastError());
Serial.print("before msg sent, returnCode: ");
Serial.println(client.returnCode());
client.publish("devices/fautest/messages/events/", messagePayload);
Serial.print("after msg sent, lastError: ");
Serial.println(client.lastError());
Serial.print("after msg sent, returnCode: ");
Serial.println(client.returnCode());

client.connect("fautest", "XXX.azure-devices.net/fautest/api-version=2018-06-30", "", false)
```

**Serial.print:**
```
12:57:04.325 -> >>>> CONNECTED to network
12:57:05.471 -> Connected
12:57:06.394 -> 
12:57:06.428 -> Attempting to connect to MQTT broker: XXX.azure-devices.net
12:57:20.236 -> You're connected to the MQTT broker
12:57:30.375 -> "T-Mobile CZ",2,11
12:57:30.444 -> {"deviceId":"fautest", "messageId":1, "yellow":0, "green":0, "cycle":0, "red":0, "reject":0}
12:57:30.444 -> getTimeStamp: 12:57:29,CET
12:57:30.479 -> gsmAccess.getLocalTime: 1575464249
12:57:30.889 -> before msg sent, lastError: 0
12:57:30.889 -> before msg sent, returnCode: 0
12:57:31.026 -> after msg sent, lastError: 0
12:57:31.026 -> after msg sent, returnCode: 0
.....
12:58:50.756 -> "T-Mobile CZ",2,10
12:58:50.789 -> {"deviceId":"fautest", "messageId":9, "yellow":0, "green":0, "cycle":0, "red":0, "reject":0}
12:58:50.789 -> getTimeStamp: 12:58:50,CET
12:58:50.824 -> gsmAccess.getLocalTime: 1575464330
12:58:51.443 -> before msg sent, lastError: 0
12:58:51.443 -> before msg sent, returnCode: 0
12:58:51.546 -> after msg sent, lastError: 0
12:58:51.546 -> after msg sent, returnCode: 0
12:59:00.837 -> "T-Mobile CZ",2,10
12:59:00.903 -> {"deviceId":"fautest", "messageId":10, "yellow":0, "green":0, "cycle":0, "red":0, "reject":0}
12:59:00.903 -> getTimeStamp: 12:59:00,CET
12:59:00.937 -> gsmAccess.getLocalTime: 1575464340
12:59:01.447 -> before msg sent, lastError: 0
12:59:01.447 -> before msg sent, returnCode: 0
12:59:06.873 -> after msg sent, lastError: -1
12:59:06.873 -> after msg sent, returnCode: 0
12:59:06.907 -> Attempting to connect to MQTT broker: XXX.azure-devices.net

```

salvq, 2019-12-04:
I just increased buffer size and msg keeps getting delivered (10+ no issue so far)...why every 10th message gets failed without buffer increase settings ? Why not 9th or 11th ?

`MQTTClient client(256);`

Thanks

256dpi, 2020-10-18:
Closing as outdated. Please reopen if problem persists.

closed 2020-10-18 by 256dpi