256dpi/arduino-mqtt #186: Socket error @ mosquitto using arduino-mqtt

ld21, 2019-12-27:
Hi

I'm using your library with ESP32s. I connect to a mosquitto broker (1.5.7) using 
`client.setOptions(10,false,1000); // Keep Alive seconds, CleanSession, Timout seconds`
`client.connect(mqtt_topic, mqtt_user, mqtt_passwort);`
`mqtt_qos = 1;`
`client.subscribe(mqtt_topic,mqtt_qos);`

(I'm aware that my client ID and the topic are (on purpose) identical.)

Unfortunately, the connection is dropped very quickly by mosquitto due to a socket error:

(from the mosquitto.log:)

    1577475051: New client connected from 77.56.110.169 as ursula_tilo_EXWsMrTwsvh78TVp (c0, k10, u'herz-o-mat').
    1577475051: No will message specified.
    1577475051: Sending CONNACK to ursula_tilo_EXWsMrTwsvh78TVp (1, 0)
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m17, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (100 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m18, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (102 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m19, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (93 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m20, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (93 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m21, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (93 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m22, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (93 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m23, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (25 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m24, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (25 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m25, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (93 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m26, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (29 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m27, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (16 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m28, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (17 bytes))
    1577475051: Sending PUBLISH to ursula_tilo_EXWsMrTwsvh78TVp (d1, q1, r0, m29, 'ursula_tilo_EXWsMrTwsvh78TVp', ... (49 bytes))
    1577475051: Received SUBSCRIBE from ursula_tilo_EXWsMrTwsvh78TVp
    1577475051: 	ursula_tilo_EXWsMrTwsvh78TVp (QoS 1)
    1577475051: ursula_tilo_EXWsMrTwsvh78TVp 1 ursula_tilo_EXWsMrTwsvh78TVp
    1577475051: Sending SUBACK to ursula_tilo_EXWsMrTwsvh78TVp
    1577475051: Socket error on client ursula_tilo_EXWsMrTwsvh78TVp, disconnecting.

Is there any way to get more information on why there is a socket error? 
Thank you for sharing any idea.
Dani

drewsed, 2020-01-04:
Hi,

I have the same issue on Wemos D1 Mini R2 (esp8266) with ENC26J60 Ethernet Board, using UIPEthernet and arduino-mqtt. I get "Socket error" on Mosquitto log every 15 minutes. Sometimes the old connection is still op on Mosquitto server side. I also have a WizWiki W7500P Eval Board running mbed-os, which is reconnecting about once a month or less, so I think the server is fine.

I assume, you are on WiFi with your ESP32, right? 

Log:
```
1578140813: Socket error on client ESP-Client, disconnecting.
1578140827: New connection from 192.168.178.46 on port 1883.
1578140827: New client connected from 192.168.178.46 as ESP-Client (p2, c1, k10, u'ESP-Client').
1578140828: New connection from 192.168.178.46 on port 1883.
1578140828: Client ESP-Client already connected, closing old connection.
1578140828: New client connected from 192.168.178.46 as ESP-Client (p2, c1, k10, u'ESP-Client').
1578141712: Socket error on client ESP-Client, disconnecting.
```
I also figured out, that a reconnect sometimes will take around 15 seconds.

My code simply publishes a reconnect-counter every few seconds and prints incoming messages to serial:
```cpp
#include <ESP8266WiFi.h> //for disabling wifi
#include <UIPEthernet.h>
#include <MQTT.h>

// ### Ethernet ###
uint8_t mac[6] = {0x00,0x01,0x02,0x03,0x04,0x05};
EthernetClient ethClient;

// ### MQTT ###
MQTTClient mqttClient;
const char mqtt_broker_addr[]= "NUC.fritz.box";
int mqtt_broker_port = 1883;
const char mqtt_client_id[] = "ESP-Client";
const char mqtt_client_user[] = "ESP-Client";
const char mqtt_client_pass[] = "";
int QoS_1 = 1;
bool noRetain = false;
const char mqtt_sub_1[] = "/ESP/in";
const char mqtt_pub_1[] = "/ESP/out";

// ### Time ###
unsigned long previousMillis;
unsigned long interval = 3000;

// ### Misc ###
int rc = 0;
int reconnectCnt = -1;

void connectMQTT() {
  Serial.print("[MQTT] connecting..");
  unsigned long stopwatch = millis();
  while (!mqttClient.connect(mqtt_client_id, mqtt_client_user, mqtt_client_pass))
  {
    Serial.print(".");
    delay(1000);
  }
  Serial.print("\n[MQTT] connected! Took ");
  Serial.print(millis()-stopwatch);
  Serial.print(" ms, ");
  Serial.print("Reconnect count: ");
  Serial.println(reconnectCnt++);
  // ### all subscriptions ###
  rc = mqttClient.subscribe(mqtt_sub_1,QoS_1);
  Serial.print("[MQTT] Subscribed, rc = "); Serial.println(rc);
  Serial.print("\t"); Serial.println(mqtt_sub_1);
}

void messageReceived(String &topic, String &payload) {
  Serial.println("[MQTT] incoming: " + topic + " - " + payload);
}

void setup() {
  // ### disable WiFi ###
  WiFi.mode( WIFI_OFF );
  WiFi.forceSleepBegin();

  // ### Serial ###
  Serial.begin(115200);
  delay(1000);

  // ### ETH ###
  rc = Ethernet.begin(mac);
    if(rc == 0){Serial.print("\n\n[ETH] Error: ");Serial.println(rc);return;}
    else Serial.println("\n\n[ETH] configuration using DHCP success");

  // ### MQTT ###
  mqttClient.begin(mqtt_broker_addr, mqtt_broker_port, ethClient);
  mqttClient.onMessage(messageReceived);
  Serial.println("[MQTT] client configured");
  connectMQTT();
  
  previousMillis = millis();
}

void loop() {

  mqttClient.loop();

  if (!mqttClient.connected()) {
    connectMQTT();
  }
  
  if(millis() - previousMillis > interval) {
    publishData();
    previousMillis = millis();
  }
}

void publishData() {
  char msgBuffer[100];
  sprintf(msgBuffer,"reconn = %i",reconnectCnt);
  rc = mqttClient.publish(mqtt_pub_1, msgBuffer, noRetain, QoS_1);
  //Serial.print("[MQTT] published, rc = "); Serial.print(rc); Serial.print(" Reconnect count: "); Serial.println(reconnectCnt);
}
```

Using QoS 0 i do not encounter this issue within my testing interval of about 2 hours.

256dpi, 2020-01-07:
Could elaborate on that happens on the Arduino side? At which point is the connection lost? And what are the values of `client.lastError()`.

drewsed, 2020-01-07:
Thanks for the reply, i didn't see this function somehow.

For me `client.lastError()` 99% returns `LWMQTT_MISSING_OR_WRONG_PACKET` if an error occurs.

And it is returned after `client.connected()` failed. Which is called right after `client.loop()`, which in turn retuns `LWMQTT_SUCCESS`

But after some time, with no change to the ESPboard (except reboot)  the interval of the failure is not periodically every 15mins anymore, but random between 10 and 240mins.

I actually think this problem is caused by UIPEthernet (which is not the most stable Library in my feelings) instead of MQTT.

drewsed, 2020-01-08:
Ok, I moved to other hardware because of several other reasons and on the OLIMEX ESP32-PoE this library works without any connection problems for about 24 hours.

Thank you for this Library!

256dpi, 2020-01-09:
@drewsed I'm glad to hear that!

ld21, 2020-01-12:
I get "-1" from client.lastError(); from the following code:

`void loop() {
  
  client.loop();
  delay(10);  // <- fixes some issues with WiFi stability

  if (!client.connected()) {
    Serial.print("Ich bin nicht mit dem MQTT-Server verbunden aus Grund «");
    Serial.print(client.lastError());
    Serial.println("» und probiere mich jetzt wieder zu verbinden.");
    connect();
  }
}`

Output from Terminal:
`16:25:16.900 -> connecting to the mosquitto broker www.herz-o-mat.ch:8883 as Client with ID pati_y49kNtKX7HTUqxDS and QoS of 1...connected!
16:25:17.862 -> Ich bin nicht mit dem MQTT-Server verbunden aus Grund «-1» und probiere mich jetzt wieder zu verbinden.
16:25:17.862 -> checking wifi...ok.https://www.herz-o-mat.ch/check_server.php sagt: &#10003;
16:25:19.518 -> Server ist erreichbar
16:25:19.518 -> `

mosquitto.log says repeatedly at the same time:
`1578842804: New connection from 77.56.100.48 on port 8883.
1578842804: Socket error on client pati_y49kNtKX7HTUqxDS, disconnecting.
1578842804: New client connected from 77.56.100.48 as pati_y49kNtKX7HTUqxDS (c0, k10, u'herz-o-mat').
1578842804: No will message specified.
1578842804: Sending CONNACK to pati_y49kNtKX7HTUqxDS (1, 0)
1578842804: Sending PUBLISH to pati_y49kNtKX7HTUqxDS (d1, q1, r0, m26, 'pati_y49kNtKX7HTUqxDS', ... (110 bytes))
1578842804: Sending PUBLISH to pati_y49kNtKX7HTUqxDS (d1, q1, r0, m27, 'pati_y49kNtKX7HTUqxDS', ... (169 bytes))
1578842804: Sending PUBLISH to pati_y49kNtKX7HTUqxDS (d1, q1, r0, m28, 'pati_y49kNtKX7HTUqxDS', ... (122 bytes))
1578842804: Sending PUBLISH to pati_y49kNtKX7HTUqxDS (d1, q1, r0, m29, 'pati_y49kNtKX7HTUqxDS', ... (108 bytes))
1578842804: Sending PUBLISH to pati_y49kNtKX7HTUqxDS (d1, q1, r0, m30, 'pati_y49kNtKX7HTUqxDS', ... (24 bytes))
1578842804: Sending PUBLISH to pati_y49kNtKX7HTUqxDS (d1, q1, r0, m31, 'pati_y49kNtKX7HTUqxDS', ... (9 bytes))
1578842804: Sending PUBLISH to pati_y49kNtKX7HTUqxDS (d1, q1, r0, m32, 'pati_y49kNtKX7HTUqxDS', ... (14 bytes))
1578842804: Sending PUBLISH to pati_y49kNtKX7HTUqxDS (d1, q1, r0, m33, 'pati_y49kNtKX7HTUqxDS', ... (43 bytes))
1578842804: Received SUBSCRIBE from pati_y49kNtKX7HTUqxDS
1578842804: 	pati_y49kNtKX7HTUqxDS (QoS 1)
1578842804: pati_y49kNtKX7HTUqxDS 1 pati_y49kNtKX7HTUqxDS
1578842804: Sending SUBACK to pati_y49kNtKX7HTUqxDS
1578842805: Socket error on client pati_y49kNtKX7HTUqxDS, disconnecting.`

Does the error "-1" help you in any way?
Thank you very much,

Dani

drewsed, 2020-01-12:
see here: [lwmqtt.h](https://github.com/256dpi/arduino-mqtt/blob/master/src/lwmqtt/lwmqtt.h)

Error `-1` says `LWMQTT_BUFFER_TOO_SHORT`.

So try sending shorter messages or [increase the buffer](https://github.com/256dpi/arduino-mqtt/issues/181).

ld21, 2020-01-12:
I increased the buffer and it solved my mqtt-problems!
Thank you Drewsed for your great advice!
Dani

fer-g3, 2020-03-23:
i'm having a lot of troubles to connect arduino with mosquitto broker on a lan network using this library, I also tried with PubSubClient and it worked ok, but every time I use this library to connect to a broker on (192.168.1.2) it send an error message -3 (LWMQTT_NETWORK_FAILED_CONNECT = -3).

`void connect(){
  clientid = String(random(0xffff), HEX);  
  while(!client.connected()){
     Serial.println("Conectando...");
     if(client.connect(clientid.c_str())){
      Serial.println("\nconnected");
      client.setWill(willTopic, willPayload, willRetain, willQos);
      client.setOptions(keepAlive, true, 3000); 
      //subscriptions
      client.subscribe(topic, qos); 
     }else{
        Serial.print("Failed: ");
        Serial.print( client.lastError());
        delay(5000);
     }
  } 
}`

**this is setup code**
`void setup() {
  Serial.begin(9600);
  Ethernet.begin(mac, local);
  delay(3000);
  client.begin(broker,1883, net);
  client.onMessage(onMessageReceived); 
}`

Please I hope anyone could help me.

closed 2020-10-18 by 256dpi