256dpi/arduino-mqtt #188: Different behavior than pubsubclient. Disconnected after publishing a msg

ortegafernando, 2020-02-02:
Hi, first of all, thanks a lot for this great library. You must be proud of it, as it improves a lot the pubsubclient library, as it can make all QOS.

I am trying to use this library with tinyGSM library with an Arduino Pro Mini. But I have found a problem. In order to know if it is from MQTT library or "client" library, I have use this sketch.

It is a tinyGSM library's example (using pubsubclient library) modified to use it with pubsubclient or with this great library. (only comment or uncomment first line)

```
//#define PUBSUB  //use PubSubClient

// Select your modem:
#define TINY_GSM_MODEM_SIM800
//#define TINY_GSM_MODEM_SIM900

// See all AT commands, if wanted
//#define DUMP_AT_COMMANDS

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
#include <SoftwareSerial.h>
SoftwareSerial SerialAT(8, 7); // RX, TX

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// Range to attempt to autobaud
#define GSM_AUTOBAUD_MIN 2400
#define GSM_AUTOBAUD_MAX 9600

// Add a reception delay - may be needed for a fast processor at a slow baud rate
//#define TINY_GSM_YIELD() { delay(2); }

// Define how you're planning to connect to the internet
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials. Leave empty, if missing user or pass
const char apn[]  = "orangeworld";
const char gprsUser[] = "";
const char gprsPass[] = "";

// MQTT details
const char* broker = "xxxxxxx.com";
const unsigned int port = 1883;

const char* topicLed = "gate/GsmClientTest/led";
const char* topicInit = "gate/GsmClientTest/init";
const char* topicLedStatus = "gate/GsmClientTest/ledStatus";

#include <TinyGsmClient.h>
#ifdef PUBSUB
  #include <PubSubClient.h>
#else
  #include <MQTT.h>
#endif

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

TinyGsmClient client(modem);
//TLS TinyGsmClientSecure client(modem);
[... 417 lines cut ...]
gate/GsmClientTest/lastWill : msg : Object
{ topic: "gate/GsmClientTest/lastWill", payload: "last will sent", qos: 0, retain: false, _topic: "gate/GsmClientTest/lastWill" … }
2/2/2020 21:56:23node: 1f52d5a1.07317a
gate/GsmClientTest/init : msg : Object
{ topic: "gate/GsmClientTest/init", payload: "GsmClientTest started", qos: 0, retain: false, _topic: "gate/GsmClientTest/init" … }
2/2/2020 21:56:30node: 1f52d5a1.07317a
gate/GsmClientTest/led : msg : Object
{ topic: "gate/GsmClientTest/led", payload: "1580676990278", qos: 2, retain: false, _topic: "gate/GsmClientTest/led" … }
2/2/2020 21:56:32node: 1f52d5a1.07317a
gate/GsmClientTest/ledStatus : msg : Object
{ topic: "gate/GsmClientTest/ledStatus", payload: "0", qos: 0, retain: false, _topic: "gate/GsmClientTest/ledStatus" … }
2/2/2020 21:56:33node: 1f52d5a1.07317a
gate/GsmClientTest/lastWill : msg : Object
{ topic: "gate/GsmClientTest/lastWill", payload: "last will sent", qos: 0, retain: false, _topic: "gate/GsmClientTest/lastWill" … }
2/2/2020 21:56:37node: 1f52d5a1.07317a
gate/GsmClientTest/init : msg : Object
{ topic: "gate/GsmClientTest/init", payload: "GsmClientTest started", qos: 0, retain: false, _topic: "gate/GsmClientTest/init" … }
2/2/2020 21:56:53node: 1f52d5a1.07317a
gate/GsmClientTest/lastWill : msg : Object
{ topic: "gate/GsmClientTest/lastWill", payload: "last will sent", qos: 0, retain: false, _topic: "gate/GsmClientTest/lastWill" … }
```

I don't know how to begin, and where to check this problem. It seems that is not because tinyGSM, as pubsubclient works well. I want to use this library as it is better than old pubsubclient, but... does anybody knowhow to solve this?

Thanks a lot.

256dpi, 2020-02-03:
@ortegafernando Thanks for writing such a detailed report!

The MQTT QoS 1/2 flows require the sending and receiving of acks. Therefore, publishing a message in the `mqttCallback` may cause a deadlock if the received message and sent message both have a non-zero QoS. I generally recommend to not send messages in the callback at all and instead flag a global variable to send it in the loop or add it to some sort of queue.

Also, depending on the payload you might want to increase the buffer sizes: `MQTTClient client(256);`.

Let me know it this helped.

ortegafernando, 2020-02-03:
Hi @256dpi , thanks a lot for your quick answer.

I am sending QOS 0. But I have tried what you say, so I have changed my sketch and add a global boolean variable to know when to publish status message.

SO I have changed loop this way:
```
  mqtt.loop();
  if (msg4ledStatus) {
      mqtt.publish(topicLedStatus, ledStatus ? "1" : "0");
      //mqtt.publish(topicLedStatus, ledStatus ? "1" : "0", false, 2);
      msg4ledStatus = false;
  }
```

And call back function:
```
      //200203: It is better NOT to publish anything inside a Callback method
      //mqtt.publish(topicLedStatus, ledStatus ? "1" : "0");
      //mqtt.publish(topicLedStatus, ledStatus ? "1" : "0", false, 2);
      msg4ledStatus = true;
```

I have uploaded sketch and it is more stable. I have done XXXXX test:
1. in the first one, after the first msg sent from node red I get a error code = -6. After that, I have sent several messages and everything goes ok. The led has alwyas switched off/on.

```
[93822] ### Available: 0 on 1
[94105] ### Got Data: 1
[94232] ### Available: 39 on 1
[94363] ### Available: 39 on 1
[94576] ### READ: 39 from 1
Message arrived [gate/GsmClientTest/led]: 1580751377460
[94861] ### Available: 0 on 1
[95414] ### Available: 0 on 1
=== MQTT NOT CONNECTED ===  lastError - returnCode: -6 - 0
Connecting to XXXXXX.com[97079] ### Available: 0 on 1
[97380] ### Got Data: 1
[97509] ### Available: 4 on 1
```
2. second test (same sketch): everything ok
3. third test (same sketch): everything ok.

So I suppose that the first time was a failure in gsm connection.

So..... thanks a lot!!! I will tell tinyGSM's developer to know about this, or a PR with the example with your library.

Next step.... check QOS 1 and QOS 2.

Thanks a lot. I close the issue. Thanks.

closed 2020-02-03 by ortegafernando

256dpi, 2020-02-03:
I'm glad it works now! I'll update the examples to indicate this more clearly.