256dpi/arduino-mqtt #190: Different behavior than pubsubclient. Disconnected after second msg

ortegafernando, 2020-02-05:
HI, I go on checking this great library.  Now I have checked to publish a msg every 60 seconds, and compare with pubsubclient (just in case).

I get an strange behaviour with arduino-mqtt and I don't know where I have to start or continue with the debuging process.

My configuration is an Arduino Pro Mini + SIM800L + TinyGSM library.
Suscribe and publish are QOS 0. And node red is also conifgured with QOS 0

This is my sketch (it is almost the usual example). As you could see I have already increased buffer to 256 (just in case, because with 128 has to be enough). Also I am using F("") in string to increase free ram in arduino (just in case)

```
/**************************************************************
 *
 *   https://github.com/256dpi/arduino-mqtt
 *
 * TinyGSM Getting Started guide:
 *   https://tiny.cc/tinygsm-readme
 *
 **************************************************************/

#define PUBSUB  //use PubSubClient

// Select your modem:
#define TINY_GSM_MODEM_SIM800
//#define TINY_GSM_MODEM_SIM900

// See all AT commands, if wanted
//#define DUMP_AT_COMMANDS

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
#include <SoftwareSerial.h>
SoftwareSerial SerialAT(8, 7); // RX, TX

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// Range to attempt to autobaud
#define GSM_AUTOBAUD_MIN 2400
#define GSM_AUTOBAUD_MAX 9600

// Add a reception delay - may be needed for a fast processor at a slow baud rate
//#define TINY_GSM_YIELD() { delay(2); }

// Define how you're planning to connect to the internet
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials. Leave empty, if missing user or pass
const char apn[]  = "orangeworld";
const char gprsUser[] = "";
const char gprsPass[] = "";

// MQTT details
const char* broker = "XXXXXX.com";
const unsigned int port = 1883;

//Change this to avoid in test overflow in buffer
const char* topicLed = "gate/led";
const char* topicInit = "gate/init";
const char* topicLedStatus = "gate/ledStatus";
const char* topicVcc = "gate/vcc";

#include <TinyGsmClient.h>
#ifdef PUBSUB
  #include <PubSubClient.h>
[... 2038 lines cut ...]
5/2/2020 20:45:24node: 1f52d5a1.07317a
gate/ledStatus : msg : Object
{ topic: "gate/ledStatus", payload: "0", qos: 0, retain: false, _topic: "gate/ledStatus" … }
5/2/2020 20:46:46node: 1f52d5a1.07317a
gate/vcc : msg : Object
{ topic: "gate/vcc", payload: "3846", qos: 0, retain: false, _topic: "gate/vcc" … }
5/2/2020 20:46:50node: 1f52d5a1.07317a
gate/led : msg : Object
{ topic: "gate/led", payload: "1580932009840", qos: 0, retain: false, _topic: "gate/led" … }
5/2/2020 20:46:52node: 1f52d5a1.07317a
gate/ledStatus : msg : Object
{ topic: "gate/ledStatus", payload: "1", qos: 0, retain: false, _topic: "gate/ledStatus" … }
5/2/2020 20:48:17node: 1f52d5a1.07317a
gate/vcc : msg : Object
{ topic: "gate/vcc", payload: "3851", qos: 0, retain: false, _topic: "gate/vcc" … }
5/2/2020 20:49:46node: 1f52d5a1.07317a
gate/lastWill : msg : Object
{ topic: "gate/lastWill", payload: "die", qos: 0, retain: false, _topic: "gate/lastWill" … }
5/2/2020 20:49:51node: 1f52d5a1.07317a
gate/init : msg : Object
{ topic: "gate/init", payload: "started", qos: 0, retain: false, _topic: "gate/init" … }
5/2/2020 20:50:07node: 1f52d5a1.07317a
gate/lastWill : msg : Object
{ topic: "gate/lastWill", payload: "die", qos: 0, retain: false, _topic: "gate/lastWill" … }
```

256dpi, 2020-02-05:
The `lastError: -6` (LWMQTT_NETWORK_FAILED_WRITE) indicates that there was an issue while writing something. Since you're printing the error after `client->connected()` we do not know which call caused the error. Most methods return a boolean to indicate the calls success. If it failed you should check `lastError` immediately and print it. Then we know which call was not successful.

Teddyz, 2020-02-05:
It might be obvious, but it would be great if you could add in the documentation or in an example if 1 means sent OK or if 0 means "No Error". 

Is it possible to get the error-texts from the library? (Or would that consume a lot of ram?)

256dpi, 2020-02-06:
In all cases `true` means success. I updated the docs accordingly.

I'm playing with the idea of adding a `client.debug()` function that prints information about whats going on under the hood. But this will consume a lot of bytes for the strings, so I'm not sure what's the best strategy here. Any ideas?

Teddyz, 2020-02-06:
I saw this that might be a good approach:
[image]

The code, in WiFiClientSecureBearSSL.cpp,  seems simple (I omitted most of the cases):
```C++
// Returns an error ID and possibly a string (if dest != null) of the last
// BearSSL reported error.
int WiFiClientSecure::getLastSSLError(char *dest, size_t len) {
  int err = 0;
  const char *t = PSTR("OK");
  if (_sc || _sc_svr) {
    err = br_ssl_engine_last_error(_eng);
  }
  if (_oom_err) {
    err = -1000;
  }
  switch (err) {
    case -1000: t = PSTR("Unable to allocate memory for SSL structures and buffers."); break;
    case BR_ERR_BAD_PARAM: t = PSTR("Caller-provided parameter is incorrect."); break;
    case BR_ERR_BAD_STATE: t = PSTR("Operation requested by the caller cannot be applied with the current context state (e.g. reading data while outgoing data is waiting to be sent)."); break;
    case BR_ERR_OVERSIZED_ID: t = PSTR("ServerHello contains a session ID which is larger than 32 bytes."); break;
    case BR_ERR_BAD_CIPHER_SUITE: t = PSTR("Server wants to use a cipher suite that we did not claim to support. This is also reported if we tried to advertise a cipher suite that we do not support."); break;
    case BR_ERR_X509_DN_MISMATCH: t = PSTR("Issuer/Subject DN mismatch in the chain."); break;
    case BR_ERR_X509_WEAK_PUBLIC_KEY: t = PSTR("Public key found in certificate is too small."); break;
    case BR_ERR_X509_NOT_TRUSTED: t = PSTR("Chain could not be linked to a trust anchor."); break;
    default: t = PSTR("Unknown error code."); break;
  }
  if (dest) {
    strncpy_P(dest, t, len);
    dest[len - 1] = 0;
  }
  return err;
}`

ortegafernando, 2020-02-10:
Hi, I have been doing some test.

With these options:   mqtt.setOptions(10, true, 15000);

- With 2400 baud rate with SIM800L: after 2 msg, it disconnects from broker
- 4800: Ok

(I was used to 2400 in my old GSM SIM800 library, with TinyGSM it seems that it is better 4800)

So I will close this issue, but first: what about this?: 

May be in future I will try AltSoftSerial or NeoICserial or NeoSWserial. Does anybody try them? Worth it over software serial?

Thanks.

closed 2020-03-08 by ortegafernando