256dpi/arduino-mqtt #191: Heap memory leak caused by client.connect() ?

robert-rudolph, 2020-02-06:
I call the following function periodically on an ESP32. The free heap decreases each time I call `connectToMQTT()`, indicating a memory leak. 

If I comment out the line containing `mqtt_client.connect()`, then the memory leak goes away, and I have perfect heap preservation. 

I've dug through the library source here and not found any unmatched `malloc()` calls, so I'm stumped as to where the memory is going. Any help would be appreciated!

```
void connectToMQTT()
{
  int retries = 0;
  int success = false;
  while ( !success && retries < 10) {
    // mqtt_client.disconnect();
    success = mqtt_client.connect("0031");
    delay(100);
    retries++;
  }
}
```

256dpi, 2020-02-06:
Do you know by how much the heap increases on every call?

robert-rudolph, 2020-02-06:
I'm loosing between 108 and 836 bytes of heap per call.

this may be a problem with the WiFiClient library for ESP32, and not this library. 
Seems some similar leak was present on ESP8266 with PubSubClient library on reconnects:
https://github.com/esp8266/Arduino/issues/4497

robert-rudolph, 2020-02-06:
Closing, this seems to be a leak in the WiFiClient or WiFiClientSecure libraries within ESP32-Arduino Core. Thank you.

closed 2020-02-06 by robert-rudolph

dmarc1234, 2022-01-05:
I am experiencing the same issue with a failed mqtt_client.connect() on the ESP32, losing 4916 bytes of heap on each call.

robert-rudolph, 2022-01-05:
I traced this to the WifiClient core, it is not an issue in the ardino-mqtt repo.

See this thread for a temp fix: https://github.com/espressif/arduino-esp32/issues/3808#issuecomment-610073056