256dpi/arduino-mqtt #196: Help needed - ISR onMessageAdvanced

AlesSt, 2020-03-31:
Hi, 
first of all thank you for great Lib.
But.....always a but :) ... as i expanded example i ran into a problem:

Setup:
ESP32 Wrover
Lib version 2.4.7
VScode

if anything else needed please let me know.

So I am using ISR onMessageAdvanced (i just hate Strings) and since i have too much stuff going in there i wanted to just raise a flag and deal with received msg after ISR but am a bit stuck here.

From MQTT i get a JSON: {"key1": 1, "key2": "value2", "key3": "true"}

and am using ArduinoJSON 6.14.1 lib to  parse it and deal with data.

But if I raise just a flag in ISR where can i pull data like Payload and topic from? Or should i copy them to a separate var within ISR? 

somwhere i read that ISRs on ESP32 should have IRAM_ATTR but it means i cant use heap as heap external ram? 
But then vars should be static volatile?

BR
Ales

AlesSt, 2020-03-31:
So now i changed ISR to raise flag that new msg from MQTT available and i copied payload to a new var but if var is already loaded in mem (from some lib procedure) i d like to extract and deal with it there and not from a copy of it (dont want duplicating it).

So main question is........where else can i get the content of Mqtt msg just arrived?

Or 

can i work without ISR and only in loop cheking if ur var(arrived Mqtt msg) has changed and if so then i process it ...... sure this works only when subscribed to only one topic.......

ALSO 
is there a way that ISR doesnt interrupt flash operation or is this only if i add ISR to IRAM?

256dpi, 2020-03-31:
First of all, the callback is not an interrupt, hence it should not be flagged that way. The callback is always called during `client.loop()` or sometimes `client.publish`, `client.subscribe` and `client.unsubscribe`. And these calls should definitely not be called during an interrupt.

Secondly, if you're not processing the data right away, it must be copied. The underlying buffer might immediately be reused after the callback returns. You either use another buffer in which you can copy the data into, or you process the JSON and only copy the things you need.

AlesSt, 2020-04-03:
Ok.......but adding IRAM_ATTR to callback stopped reboots saying using cache and cache disabled ......core 1 in ISR mode something like that.....must be some other procedure then.......yes i did that....put a flag up copied payload (as subscription to one topic ONLY) and after client.loop() i parse payload.......but changing callback to just raise flags and copy payload didnt help.

Need to dismantle other libs that i have......as said thank you for the effort and help :)

BR and stay safe :)

closed 2020-04-03 by AlesSt