256dpi/arduino-mqtt #197: ESP32DevelopmentBoardSecure example has LWMQTT_BUFFER_TOO_SHORT error

BrianSidebotham, 2020-04-18:
The ESP32DevelopmentBoardSecure example sketch for the ESP32 fails to receive the first message after connect and sits in an endless re-connecting loop.

Inserting the `getLastError()` call before the re-connect shows an error of `-1` (`LWMQTT_BUFFER_TOO_SHORT`) - in order to get the example to run I have to initialise the client with a much larger buffer: `MQTTClient client(1024);`

If the buffer is smaller than this, the example doesn't work for me.

The first message I receive is always bonkers accordng to the serial output:

```
checking wifi....
connecting...
connected!
incoming: /hello - 01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
incoming: /hello - world
incoming: /hello - world
```

256dpi, 2020-04-20:
The first message you see is from someone that published a retained message in the "try" namespace on shiftr.io. Sure, not an ideal experience, but theres nothing we can do about it... 🤷‍♂️

BrianSidebotham, 2020-04-20:
Perhaps we can immediately publish a blank retained message (The topic can only have a single retained message) and that should prevent us from receiving the "bad" retained message that requires a large buffer.

I think a simple `client.publish("/hello", "", true);` should make the example code more reliable for people.

I can do a PR if you think this is acceptable?

256dpi, 2020-04-23:
Yes, let's do that. The `client.publish("/hello", "", 1, true)` should be placed right before the subscribe to ensure the retained message has been reset before subscribing. Also a comment should explain why it is needed and that the program may still crash if the "adversary" is online and sending. Would you be able to update all examples?

256dpi, 2023-02-04:
Closing in favour of #199.

closed 2023-02-04 by 256dpi