256dpi/arduino-mqtt #199: Skip Large Messages

256dpi, 2020-04-29:
If a client receives a message that is bigger than its receive buffer if will crash and return the error "LWMQTT_BUFFER_TOO_SHORT". For this library the better approach would be to just skip and log large messages.

maglevize, 2020-09-25:
can we add the size for this buffer?

Sisco21, 2021-02-11:
Hello, i had the same issue. Did you modified the behavior? Can we catch the error?

mrdc, 2021-09-20:
@256dpi 

Hello, 
Nice feature to have - on large payloads device loses the connection to MQTT broker, so we have to protect it from this kind of possible attack.

256dpi, 2023-02-04:
We should expose `lwmqtt_drop_overflow` to skip too large messages.

256dpi, 2023-02-05:
Added in eae214b1c2ca0d861037617a0ed0128e3c5ec306.

closed 2023-02-05 by 256dpi