256dpi/arduino-mqtt #202: Connection still running if broker is down

catalinroweb, 2020-05-06:
Hello,

I'm using an MKR1000 board connected to my own mqtt broker.
I create the connection between board and broker, then i shut down the broker. The board returns even after 5 minutes `true` when i call `.connected()`.

This is an error or should i set a timeout?

256dpi, 2020-05-06:
TCP connections can stay "connected" falsely for a very long time if no data is shared. For that purpose MQTT has the keep alive ping/pong mechanism.

catalinroweb, 2020-05-06:
Thank you for your quick reply!

I've tried now to set `client.setOptions(10, true, 1000);` but the result is the same. The board send at every 5 seconds a ping, so data is still shared in my opinion.

256dpi, 2020-10-18:
@catalinroweb Any updates on this?

256dpi, 2020-11-14:
Closing as outdated. Please reopen of problems still exists.

closed 2020-11-14 by 256dpi