256dpi/arduino-mqtt #203: disconnects milisecond after connect

kayleejacob1, 2020-05-07:
Issue I am have is that it disconnect after it connects I am able to get one message out but only have few millisecond before it disconnects again

kayleejacob1, 2020-05-07:
if you see something is calling stop socket

[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
AWS not connected
Connecting to AWS IOT
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 200288
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:115] start_ssl_client(): Loading CA cert
[V][ssl_client.cpp:163] start_ssl_client(): Loading CRT cert
[V][ssl_client.cpp:170] start_ssl_client(): Loading private key
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
[D][ssl_client.cpp:208] start_ssl_client(): Protocol is TLSv1.2 Ciphersuite is TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
[D][ssl_client.cpp:210] start_ssl_client(): Record expansion is 29
[V][ssl_client.cpp:216] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:225] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:240] start_ssl_client(): Free internal heap after TLS 155016
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
AWS IoT Connected!
[V][ssl_client.cpp:279] send_ssl_data(): Writing HTTP request...
sucess

kayleejacob1, 2020-05-08:
so its not milliseconds but about 15 seconds if I send message faster then there is no issue

Teddyz, 2020-05-08:
You have client.loop(); in the loop?

kayleejacob1, 2020-05-11:
not in my main loop no should I

Teddyz, 2020-05-11:
I think that is worth trying.
[image]

closed 2020-10-18 by 256dpi