256dpi/arduino-mqtt #204: "messageReceived" not working when inserting RTC timestamp in loop?

MrSiO, 2020-05-12:
Hi,
I'm using the MQTT.h library with success in one of my project using Adruino UNO w/Ethershield based on your ArduinoEthernetShield.ino exemple.

I wanted to add to the loop a timestamp based on RTC clock every second. _(DS3231 - Tested and working)_
Once the RTC commands is added to the "if (millis() - lastMillis > 1000)" time gets displayed every seconds, but the reception of MQTT messages stops working.
As if the "void messageReceived(String &topic, String &payload)" is not triggered anymore by incoming messages.

I can confirm the arduino is connected to broker and subscribe to topic.
Thx for any helpfull infos.

Here is my code sample:

```
#include <Ethernet.h>
#include <MQTTClient.h>
#include <MQTT.h>
#include <Wire.h>
#include "RTClib.h"
RTC_DS3231 rtc;

...
...
...

void loop() { 
  client.loop();                                              //MQTT INFINITE LOOP WATCH FOR INCOMING MESSAGES
  if (!client.connected()) {                                  //IF CLIENT GETS DISCONNECTED
    Serial.print("\nNOT CONNECTED TO BROKER....");
    connect();                                                //RECONNECT
  }
// publish a message roughly every second.
  if (millis() - lastMillis > 1000) {
    lastMillis = millis();
    Serial.print("\n--------- DEBUG 6 ------------\n");
    DateTime now = rtc.now();                                //THIS CAUSE ISSUE
    Serial.print(now.hour(), DEC);                           //THIS CAUSE ISSUE
    Serial.print(':');                                       //THIS CAUSE ISSUE
    Serial.print(now.minute(), DEC);                         //THIS CAUSE ISSUE
    Serial.print(':');                                       //THIS CAUSE ISSUE
    Serial.print(now.second(), DEC);                         //THIS CAUSE ISSUE
    Serial.println();                                        //THIS CAUSE ISSUE
  }
}
```

256dpi, 2020-05-13:
I think you accidentally removed the `client.publish()` call within the if statement?

MrSiO, 2020-05-13:
There is no need to publish at this point. It's only listening on topic and reacts to incoming topic/payload.

Problem seems to be related to I2C and maybe interrupts... 

Looks like the I2C interfere with the MQTT incoming message interrupt...  Is this possible?

256dpi, 2020-10-18:
Closing as outdated. Please reopen if problem persists.

closed 2020-10-18 by 256dpi