256dpi/arduino-mqtt #207: Google pub/sub topic subscription in esp8266

murilobaliego, 2020-06-08:
Hello, I'm just starting working with google pub/sub and I'm wondering how to use ESP8266 to subscribe to a topic to receive messages. I can send from Google Pub/Sub dashboard and message arrives with topic /devices/<dev-id>/config or /devices/<dev-id>/commands

But I'm wondering who to subscribe my ESP8266 to pull messages from a different topic like:
projects/<project-id>/topics/<topic-id>

Thanks in advance,
Murilo

256dpi, 2020-06-15:
Have you tried `client.subscribe('topic')`?

mars000, 2020-08-09:
I have same issue as Murilo - I've tried similar to what Murilo with no success. Anyone had success is subscribing to a different topic on google cloud Pub/Sub ?

256dpi, 2020-10-18:
Closing as outdated. Please reopen if problem persists.

closed 2020-10-18 by 256dpi