256dpi/arduino-mqtt #208: Pb with MKR 1010 Wifi

fmatray, 2020-06-12:
Hello,
I tried the example ArduinoWiFi101 on a MKR 1010 Wifi. I just changed the driver to WifiNina.
When receiving fast messages, the client disconnects and reconnects.

```
21:29:04.520 -> checking wifi...
21:29:04.520 -> connecting...
21:29:05.917 -> connected!
21:29:05.917 -> incoming: octoPrint/temperature/tool0 - {"_timestamp": 1591990144, "actual": 22.03, "target": 0.0}
21:29:05.917 -> incoming: octoPrint/temperature/bed - {"_timestamp": 1591990144, "actual": 21.41, "target": 0.0}
21:29:05.917 -> checking wifi...
21:29:05.917 -> connecting...
21:29:05.951 -> connected!
21:29:05.951 -> incoming: octoPrint/temperature/tool0 - {"_timestamp": 1591990144, "actual": 22.03, "target": 0.0}
21:29:05.951 -> incoming: octoPrint/temperature/bed - {"_timestamp": 1591990144, "actual": 21.41, "target": 0.0}
21:29:05.986 -> checking wifi...
21:29:05.986 -> connecting...
21:29:05.986 -> connected!
21:29:06.019 -> incoming: octoPrint/temperature/tool0 - {"_timestamp": 1591990144, "actual": 22.03, "target": 0.0}
21:29:06.019 -> incoming: octoPrint/temperature/bed - {"_timestamp": 1591990144, "actual": 21.41, "target": 0.0}
21:29:06.019 -> checking wifi...
21:29:06.019 -> connecting...
21:29:06.019 -> connected!
[...]
```

I this is the subscriptions :
```
  client.subscribe("octoPrint/temperature/#"); // tool0, tool1 and bed
  client.subscribe("octoPrint/event/#");  // Octoprint events
```

Any clue about this problem ?
Thanks a lot.

fmatray, 2020-06-14:
I tried with the example in wiki for the MKR 1010, including the delay(1000).

I want to subscribe to several topics
```
  client.subscribe("octoPrint/temperature/#");  // 2 topics tool0 and bed
  client.subscribe("octoPrint/event/PrinterStateChanged");
  client.subscribe("octoPrint/event/PowerOn");
  client.subscribe("octoPrint/event/PowerOff");
  client.subscribe("octoPrint/progress/printing");
```
I tried several configurations :
- the temperature only : it works as long as it receive frequent datas. Else I get disconnections every loop.
```
15:52:17.406 -> connected to WiFi!
15:52:17.406 -> 
15:52:17.406 -> connecting to broker...
15:52:17.444 -> Connected to MQTT
15:52:17.444 -> incoming: octoPrint/temperature/tool0 - {"_timestamp": 1592142736, "actual": 24.06, "target": 0.0}
15:52:17.444 -> incoming: octoPrint/temperature/bed - {"_timestamp": 1592142736, "actual": 44.71, "target": 0.0}
15:52:18.796 -> checking wifi...
15:52:18.796 -> connected to WiFi!
15:52:18.796 -> 
15:52:18.796 -> connecting to broker...
15:52:18.796 -> Connected to MQTT
15:52:18.796 -> incoming: octoPrint/temperature/tool0 - {"_timestamp": 1592142738, "actual": 23.59, "target": 0.0}
15:52:18.796 -> incoming: octoPrint/temperature/bed - {"_timestamp": 1592142736, "actual": 44.71, "target": 0.0}
15:52:19.924 -> checking wifi...
15:52:19.924 -> connected to WiFi!
```

- If I try to add another topic, it loops for ever : Connecting, retrieving the same MQTT message and disconnection.

256dpi, 2020-06-15:
Is it possible the message is bigger than the default buffer size of 128 bytes?

fmatray, 2020-06-20:
Yes and the retain flag

closed 2020-06-20 by fmatray