256dpi/arduino-mqtt #211: Core panic when used in GPIO interupt

Matthijsy, 2020-06-28:
I am currently using this library to send a message to MQTT when I get some input change from an GPIO input. I set this up using the interupts from GPIO. However, when I then call `client.publish` it results in a core panic. Any idea how this can be resolved?

This is the used code:
```
#include <WiFi.h>
#include "driver/gpio.h"
#include <MQTT.h>

#define SENSOR_GPIO GPIO_NUM_4

const char *ssid = "****";
const char *password = "****";
const char *mqtt_server = "192.168.1.4";

WiFiClient wifiClient;
MQTTClient client(512);

static void gpio_change(void* arg) {
  client.publish("topic", "0", false, 0);
}

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }

  client.begin(mqtt_server, wifiClient);
  while (!client.connect("***", "***", "***)")) {
    Serial.print(".");
    delay(1000);
  }

  gpio_set_pull_mode(SENSOR_GPIO, GPIO_PULLUP_ONLY);
  gpio_install_isr_service(0);
  gpio_set_intr_type(SENSOR_GPIO, GPIO_INTR_ANYEDGE);
  gpio_isr_handler_add(SENSOR_GPIO, gpio_change, (void*) SENSOR_GPIO);
}
```
The panic:
[image]

256dpi, 2020-06-28:
You can't publish from an interrupt. Instead, you need to store the sensor value somewhere and publish the value from the main loop.

Matthijsy, 2020-06-28:
Okay, thanks for your quick response! Then I will publish from the main loop.

closed 2020-06-28 by Matthijsy