256dpi/arduino-mqtt #216: Client disconnects when publishing QoS 2 message

shanejohnpaul, 2020-09-03:
When subscribed to a topic with QoS 2, publishing a QoS 2 message after receiving a QoS > 0 message from the subscribed topic disconnects the client from the broker. And this only happens sometimes.

This sounds confusing. What I tried is the following

1. Took ESP32DevelopmentBoard built-in example
2. Changed QoS of `subscribe` and `publish` functions to 2

This replicates this error. But if you increase the publishing interval to 2 seconds it's working. Anything I could do here to solve this issue?

I'm using Adafruit ESP32 feather.

TheMasterofBlubb, 2021-01-30:
```
void MQTT_Data_Handler::loop(void *instance_ptr)
{
    MQTT_Data_Handler *instance = (MQTT_Data_Handler *)instance_ptr;
    for (;;)
    {
        if (instance->client.connected() && instance->publish_queue.size() != 0)
        {
            Serial.println("Read Publish Queue");
            publish_entry publish_set = instance->publish_queue.front();
            instance->client.publish(publish_set.topic, publish_set.payload, publish_set.payload.length(), publish_set.QoS);
            Serial.println("Entry published");
            instance->publish_queue.pop();
        }
        Serial.println("Calling MQTT Loop");
        instance->client.loop();
        Serial.println("Called MQTT Loop");
        delay(20); // <- fixes some issues with WiFi stability

        if (!instance->client.connected())
        {
            Serial.println("Reconnect MQTT");
            instance->connect();
        }
    }
}
```

MQTT_Data_Handler is a class with some helper functions
MQTT_Data_Handler->client is the MQTT client of this library
publish_set is a struct of a topic, a payload and a QoS

What happens here is that :
A: instance->client.connected() is true when starting the execution
B: the  instance->client.publish() happens
C: instance->client.loop(); gets executed
D: instance->client.connected() is false in the Reconnect check

Reproducable at about 95% with QoS 2 , QoS 0 is working like a charm

256dpi, 2021-02-01:
@TheMasterofBlubb: Could you also test with QoS1? And, maybe you can upload a striped sketch that still shows this behaviour?

TheMasterofBlubb, 2021-02-01:
Im currently investigating a crash issue regarding the publishing and cant provide a striped version, as its not properly working in the current state (is there any known issue in calling the publish-function from a xTask?).

I will provide one as soon as i get it workin.

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>

256dpi, 2021-02-01:
As long as all tasks using the object run on the same core it should be fine.

TheMasterofBlubb, 2021-02-01:
Hmmmmm, i have a object that is holding the client and some additional stuff. Its created on runtime with a pointer from main(core 1). Then i have on another core (core0) a MQTT Loop that grabs stuff to publish from a mutexed stack (as my messages are quite big) and calls client.loop().

So technically only 1 core should access the client at all, but i get a metric ton of crashes in the lwmqtt code with (StoreProhibited), no matter the QoS.

If you like i can make a separete issue for that, as soon as i figure out why its crashing this time (yeah im having issues all around since i added MQTT to my sketch xD)

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>

TheMasterofBlubb, 2021-02-01:
Heads up:
I fixed all my Heap, dereferencing and stack overflow issues

Now i cant replictate the issue anymore, its running smooth as hell even qith QoS 2

Current setup involves a xQueue beeing filled in one thread and emptied in another.

My initial issue with lwmqtt was that i made a rookie mistake (im a professional Software engineer for a year now and coding in general for 8+ years). Using your example sketch i did client = MQTTClient(1024); what makes the buffers internally go BRRT XD

256dpi, 2021-02-01:
Good to hear it worked! FYI: In case you're using esp-idf you may find https://github.com/256dpi/esp-mqtt interesting. It integrates more natively with esp32 and ist a lot faster.

closed 2021-05-28 by 256dpi

gimbus109, 2022-11-25:
First of all - thanks for the great library.
The only trouble I've encountered is while publishing messages with QoS=2.

I do not want to add a new issue, my problem is exactly the same as described here. Maybe the problem is with my setup:
- Raspberry Pi Pico with earlephilhower core
- SIM 868 GSM Modem
- TinyGSM library by vshymanksyy

The stripped-down piece of code to reproduce the problem:

```
void loop() {
  mqtt.loop();
  if (networkConnect(GSM_TIMEOUT)) {
    if (!mqtt.connected()) {
      mqttReconnect();
    }
  }

  if (tick) {
    tick = false;
    tried++;
    bool published = mqtt.publish(topicData, mqttMessage,mqttMessageSize,false,2);

    if (published) {
      succeeded++;
      SerialMon.print("Success:  ");
    } else {
      SerialMon.print("Failed:  ");
    }
    SerialMon.print(succeeded);
    SerialMon.print("/");
    SerialMon.println(tried);
    
    SerialMon.print("Status mqtt: ");
    SerialMon.println(mqtt.connected());
  }
}

```

The result while QoS is set to 0 or 1:
```
Success:  1/1
Status mqtt: 1
Success:  2/2
Status mqtt: 1
Success:  3/3
Status mqtt: 1
```
and while  QoS is 2:
```
Failed:  0/1
Status mqtt: 0
[34894] MQTT disconnected, reconnecting
[34897] lastError :  -9 , returnCode:  0
Failed:  0/2
Status mqtt: 0
[44899] MQTT disconnected, reconnecting
[44902] lastError :  -9 , returnCode:  0
Failed:  0/3
Status mqtt: 0
```
When QoS is set to 2, mqtt.publish() always returns false. I can reproduce the above result with 100% certitude, despite the size of the message. But the message is sometimes sent (very rare) ...
Could anybody try to confirm?
Thanks a lot in advance