256dpi/arduino-mqtt #225: AWS 'provisioning by claim' support

societyofrobots, 2020-11-09:
I need to do MQTT scalably and securely for thousands of devices. Amazon AWS allows this via 'provisioning by claim', where a device starts off with a temp cert in the factory firmware, and then exchanges a new cert so that each device can securely use MQTT:
https://docs.aws.amazon.com/iot/latest/developerguide/provision-wo-cert.html#claim-based

Are there any plans to implement this, or a library you can point me to?

256dpi, 2020-11-11:
I have the feeling this is way out of scope for this project. TLS is anyway handled outside of the library as this varies heavily from platform to platform.

256dpi, 2021-02-01:
Closed due to being out of scope.

closed 2021-02-01 by 256dpi