256dpi/arduino-mqtt #231: Client keeps getting disconnected from mqtt server via timeout

DrTron, 2021-01-30:
Is there a way to check or modify the keepalive function, if there is one?
It looks like the library is not correctly issuing keepalive packets to the broker.

I keep getting disconnects and the arduino reconnects frequently.
In the broker logs this shows every 10-20 seconds or so:

`1611971712: New connection from 172.16.0.146 on port 1883.`
`1611971712: New client connected from 172.16.0.146 as esp32-2 (c1, k10).`
`1611971728: Client esp32-2 has exceeded timeout, disconnecting.`
`1611971728: Socket error on client esp32-2, disconnecting.`

The arduino mostly listens to a topic and seldom publishes.
In the code example this does not apply as a message is published every second, avoiding any timeout.

Teddyz, 2021-01-30:
I don't know if it is going to help you, but it is possible to change keepalive. See [https://github.com/256dpi/arduino-mqtt](url)

For me, running this library on a dozen ESP8266, it is very stable.

manolodd, 2021-03-13:
AFAIK keepalive and message receiving are done when mqtt.loop() is called. I had same issues and the reason was my loop() was doing lot of work and when mqtt.loop() was called at the end of loop(), keepalive was expired. Increasing keepalive solved the situation.

closed 2021-05-28 by 256dpi