256dpi/arduino-mqtt #233: timeout of connect can't be changed

weinerg, 2021-02-10:
Hello,
the timeout of connect(...) is always 5 seconds, even if I change the timeout by setTimeout(millisecs);
I want to have a shorter timeout, because I use MQTT for deep sleep sensors and the uptime should be as short as possible, even in case of errors.

Is there a possibility to change the timeout of connect ?

Thx.

256dpi, 2021-02-10:
No. Unfortunately, the Arduino API's have internal timeouts that cannot be changed.

weinerg, 2021-02-10:
Ok, so I no longer have to search for a a mistake in my code.
Many thx.

weinerg, 2021-02-11:
Actually I have fould a solution for setting MQTT timeout.
You just have to set the timeout of the wifi client afer begin:
  clientMQTT.begin(brokerIP, clientWiFi);
  clientWiFi.setTimeout(MQTT_TIMEOUT);
  while (!clientMQTT.connect(clientId) ) { ...

closed 2021-02-11 by weinerg