256dpi/arduino-mqtt #245: onMessage called every time that subscribe to a topic

alirezaimi, 2021-04-23:
Hi
When my esp8266 subscribe to topic if there is value in it, onMessage() function called ! 
Why !? is this right behavior ?

256dpi, 2021-04-26:
Every subscription to a topic will cause the last retained message to be sent to the client.

alirezaimi, 2021-04-26:
Thx, but how can I subscribe to a topic on my esp board that does not call onMessage method ?

David-2D3FE9, 2021-04-27:
Hello, 

I am not sure to understand you want to do:
* You want your ESP to suscribe to a topic of an mqtt broker 
* But you don't want the `.onMessage()` method

I am right ?

alirezaimi, 2021-04-28:
Hi, 
I want my esp to subscribe to a topic and after that subscription the .onMessage method does not called for this subscription ./
as you said before : Every subscription to a topic will cause the last retained message to be sent to the client. ,, in esp board this will cause to calling onMessage method after that subscription that in some case not important .

David-2D3FE9, 2021-04-28:
You don't need to suscribe to a topic to be able to publish somthing in it .
* Suscrire is to read a topic  
* Publish is to write in a topic  

In other words you can comment or delete `client.subscribe("/hello");` in your test script.
Doing this, wil still publish the data`world` in the `hello` topic without any return for your board

alirezaimi, 2021-05-03:
@David-2D3FE9 Hi and thanks for your reply, but when esp doesn't subscribe to a topic doesn't read message come from other side of mqtt server, i mean mqtt client, the onMessage does not work at all and i can't command to my esp from outside .

the real problem is every time that i subscribe to a topic, the onMessage called that i think this is wrong beha`vior .

256dpi, 2021-05-04:
As I suggested in my first comment, this behaviour sounds like there is a message retained on your topic that is sent every time you subscribe. That behaviour is correct and documented in the specification. Also note: This library cannot produce MQTT messages out of thin air, if onMessage is called, it's because the server has sent a message.

To your other problem: The library does not store subscriptions and resubscribe them when reconnecting. You either need to resubscribe or use a persistent MQTT session if the broker allows it.

David-2D3FE9, 2021-05-04:
maybe you can paste your code, I will try to help you
we may start a discussion I think, we will came back here if the problem persist or to close this issue   
Because as sayed by 256dpi the problem sounds being outhere
;-)

256dpi, 2021-05-28:
I'm closing this for now as the issue was probably in user code.

closed 2021-05-28 by 256dpi

alirezaimi, 2021-07-16:
There is no issue in my code, after every publish on server, the onMessage` called ! why !?
When my esp8266 publish a message to mqtt server, the onMessage get same message comes from server ! this is odd ! if there is a way that i could get message sender info i maybe could turn around issue for now but should be a better logical way to call onMessage function .