256dpi/arduino-mqtt #247: connect timeout

alirezaimi, 2021-05-03:
Hi

When my internet connection lost, my loop function run every 9 seconds because stuck in mqtt connection and publish part of code. why ? and how can pass this part if internet connection lost and doest stuck in seconds that i define somewhere ?
is there any param for connect method ?

in setup :

      ...
      ///// mqtt
      mqttclient.setOptions(5, true, 500);
      mqttclient.begin(MQTT_SERVER, MQTT_SERVERPORT, WiFiclient);
      mqttclient.onMessage(messageReceived);
      mqtt_connect();
      ...
      
      
in loop :

    ...
    mqttclient.loop();
    ...
    
    delay(1000);
    
    
and some where in my code define mqtt_connect func :

    void mqtt_connect() {
      WiFiclient.setInsecure();
      Serial.println(">> connecting to broker...");
      mqttclient.connect("MQTT-client-imi-arduino", MQTT_USERNAME, MQTT_PASSWORD);

256dpi, 2021-05-04:
This is a known problem. The common Arduino APIs do not yet expose a way to set a connect timeout. Depending on your platform you might find a way to tweak the network implementation to use a lower timeout.

alirezaimi, 2021-07-16:
sad :( 
Is setting `timeout` in this option can help me ?  `void setOptions(int keepAlive, bool cleanSession, int timeout); `

alirezaimi, 2021-07-21:
I can't find out any way because I can't understand the real problem cause my this issue :((((

256dpi, 2023-02-04:
Closing as stale. Please reopen if issue persists.

closed 2023-02-04 by 256dpi

alirezaimi, 2023-03-14:
The main problem is still exist ! and the timeout option is very important option that when the internet connection lost handle this problem and pass it right away without any delayed .

alirezaimi, 2023-03-14:
what about this parameter (timeout) :
void setOptions(int keepAlive, bool cleanSession, int timeout); 

what is this option ?