256dpi/arduino-mqtt #249: Unable to publish very large payloads (e.g. images)

plcengineer, 2021-05-22:
In preparation of data transmitting, the client copies the given payload into an internal buffer. This action causes the usage of huge memory and data redundancy.
In my case, I would send an PNG image recorded with the ESP32-Cam Module. So the payload comes from the camera memory as pointer and without the need of copying the whole frame buffer.

plcengineer, 2021-05-22:
> In preparation of data transmitting, the client copies the given payload into an internal buffer. This action causes the usage of huge memory and data redundancy.
> In my case, I would send an PNG [...]

 --> Made a possible solution in my fork of this very good library: see last commits: https://github.com/plcengineer/arduino-mqtt

256dpi, 2021-05-28:
Yes this would be nice to get fixed. Possibly we could do this together with #145. In any case this should be first implemented in [lwmqtt](https://github.com/256dpi/lwmqtt).

plcengineer, 2021-05-28:
That would be great. I could add some sort of solution for reading too. But I would need a bit more time. Until now, I looked for an hour in your code.. Maybe next week I can dive a bit deeper.

256dpi, 2023-02-05:
With the latest update of lwmqtt, message payloads are now sent directly and not written to the write buffer anymore. Therefore, very large payloads are possible now. The feature is planned to be available with the next release (2.6).

closed 2023-02-05 by 256dpi