256dpi/arduino-mqtt #260: MQTT broker username and password

Andrew-Wasserman, 2021-07-07:
Hi Joël,
May I ask your help.
I want to connect to an MQTT broker that needs a username and password.
Where in the  "mqtt.begin();" function do I put the user name and password?
My first guess was the following: 
mqtt.begin("MQTT-broker-server", "port", "wificlient", "mqtt-broker-username", "mqtt-user-password" );
However it does not seem to be working.
Could you perhaps point me in the correct direction?
Your help would be much appreciated.

CptHolzschnauz, 2021-07-20:
client.begin("username.cloud.shiftr.io", net);
client.connect("devicename", "channel_to_subscribe", "password");

256dpi, 2023-02-04:
Closing as this has been most likely solved.
@CptHolzschnauz Thanks for stepping in!

closed 2023-02-04 by 256dpi