256dpi/arduino-mqtt #264: Unwanted side effect of MQTT publish (Connection Close)

thehellmaker, 2021-09-14:
In the code here (https://github.com/256dpi/arduino-mqtt/blob/master/src/MQTTClient.cpp#L333 ) its closing the connection when publish fails. Publish can fail for multiple reasons some temporary and some permanent. In my case the failures are temporary and retrying succeeds. Closing the connection makes this really unstable. The library should only tell if the publish was successful or not (with the right error code) and it should be the responsibility of the caller to close the connection or retry publish

thehellmaker, 2021-09-14:
Please let me know if this makes sense. I will open a pr for removing this close from publish.

256dpi, 2023-02-04:
Closing for now. Please reopen if issues persists.

closed 2023-02-04 by 256dpi