256dpi/arduino-mqtt #271: Unable to connect to broker when used with Portenta H7

Twinkle299, 2021-11-09:
I have been using this library with both MKR NB1500 and MKR Zero + Ethernet Shield with a lot of success, 

I am attempting to use a Portenta H7, using EthernetClient i can access the network fine(i perform an http get etc) however using this EthernetClient with with the mqttClient it fails to connect to the broker (this same approach works perfectly with MKR boards).

Has anyone successfully used the library along with a Portenta ?

InconceivableDuck, 2023-01-11:
I ran into this problem when passing a static memory buffer to MBED's Thread constructor. When not passing a buffer, the library connects successfully. The issue isn't related to this library. Arduino's WiFiClient, for me, won't connect to anything inside a thread when using a static memory buffer.

256dpi, 2023-02-05:
@InconceivableDuck Thanks for the update! However, I'm not sure what we should do about it. We could make a precautionary call to `client->stop()` on every `connect()`, but this could trigger other bugs in network implementations that do not expect a call to `stop` before `connect`... Therefore, It seems to me that this is better fixed in MbedClient than in this library.

razvanphp, 2023-12-19:
@InconceivableDuck thanks for the tip, this solved my problem too!

```
void connect() {

  while (!mqttClient.connect("arduino")) {
    // There's an issue with MbedClient initializing the underlying
    // socket before connection is established (e.g. WiFi). Need to
    // stop on each attempt to force a new socket to be created.
    ethClient.stop();

    Serial.print(".");
    delay(1000);
  }
...
}
```

We switched from MKR 1010 + ETH shield to Portenta H7 Light + ETH shield and we noticed that broker connects at startup, but when taking the cable off and putting it back on, network comes back, but mqttClient does not reconnect.

Any suggestion on how to create an issue to get the issue fixed in [MbedClient](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/SocketWrapper/src/MbedClient.cpp#L81)? I don't really understand the details to create an issue there....

Thank you!
R