256dpi/arduino-mqtt #273: onMessage callback never called

JonasNuvibit, 2021-11-24:
Hello Guys

I use your lib to connect my esp32 to AWS IOT.
After connecting to Wifi and IOT I publish an empty message to the topic "$aws/things/chicken-door/shadow/name/door/get"
AWS IOT has some logic. So the empty message to this topic triggers a new message to the topic "$aws/things/chicken-door/shadow/name/door/get/accepted" containing a json document of the things state (they call it shadow).

Until here everything works fine. I see the empty message in the "get topic" and I see the json file in the "get/accepted topic" in the AWS console.

But unfortunately the message is never received by the ESP32. the onMessage callback function is never called.

Here is my code:

```
#include "secrets.h"
#include <WiFiClientSecure.h>
#include <MQTTClient.h>
#include <ArduinoJson.h>
#include "WiFi.h"

// Topics
#define AWS_IOT_PUBLISH_TOPIC   "$aws/things/chicken-door/shadow/name/door/get"
#define AWS_IOT_SUBSCRIBE_TOPIC "$aws/things/chicken-door/shadow/name/door/get/accepted"

int msgReceived = 0;
String rcvdPayload;
char sndPayloadOff[512];
char sndPayloadOn[512];

WiFiClientSecure net = WiFiClientSecure();
MQTTClient client;

void connectAWS()
{
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.println("");
  Serial.println("###################### Starting Execution ########################");
  Serial.println("Connecting to Wi-Fi");

  while (WiFi.status() != WL_CONNECTED){
    delay(500);
    Serial.print(".");
  }

  // Configure WiFiClientSecure to use the AWS IoT device credentials
  net.setCACert(AWS_CERT_CA);
  net.setCertificate(AWS_CERT_CRT);
  net.setPrivateKey(AWS_CERT_PRIVATE);

  // Connect to the MQTT broker on the AWS endpoint
  client.begin(AWS_IOT_ENDPOINT, 8883, net);

  // Create a message handler
  client.onMessage(messageHandler);

  Serial.println("Connecting to AWS IOT");

  while (!client.connect(THINGNAME)) {
    Serial.print(".");
    delay(100);
  }

  if(!client.connected()){
    Serial.println("AWS IoT Timeout!");
    return;
  }

  // Subscribe to a topic
  client.subscribe(AWS_IOT_SUBSCRIBE_TOPIC);

  Serial.println("AWS IoT Connected!");
}

void messageHandler(String &topic, String &payload) {
  msgReceived = 1;
  rcvdPayload = payload;
  Serial.println("something received");
}

void setup() {
  Serial.begin(115200);
  
  connectAWS();
  
  Serial.println("Get shadow");
  client.publish(AWS_IOT_PUBLISH_TOPIC, "");
  
  Serial.println("##############################################");
}

void loop() {
   if(msgReceived == 1)
    {
//      This code will run whenever a message is received on the SUBSCRIBE_TOPIC_NAME Topic
        delay(100);
        msgReceived = 0;
        Serial.print("Received Message:");
        Serial.println(rcvdPayload);
      Serial.println("##############################################");
    }
  client.loop();
}
```

Thanks in advance for looking into this.

air1x1, 2021-12-17:
I had a similar issue using Azure and 'Soracom Beam'. Strange thing was, when sent a message to the client, the client became disconnected. That said, I had no issues sending the message.

To double check if it was a configuration issue, I used a different client and it worked.
https://platformio.org/lib/show/89/PubSubClient
https://github.com/knolleary/pubsubclient/blob/master/examples/mqtt_basic/mqtt_basic.ino

256dpi, 2023-02-04:
Closing as issue is not related to the library itself.

closed 2023-02-04 by 256dpi