256dpi/arduino-mqtt #274: ESP8266 will not reconnect when using MQTT

gskjold, 2021-12-05:
If I let my device connect to MQTT and deauth the device from my WiFi, it will not reconnect. If I run the same code, but to not connect to MQTT, WiFi reconnect works just fine.

Code with the problem is here:
https://github.com/gskjold/AmsToMqttBridge/tree/dev-1.6.0

Master brach is ok, see difference here: 
https://github.com/gskjold/AmsToMqttBridge/pull/154

ESP WiFi debugging enabled, output from start:

SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with no23-han, channel 1
dhcp client start...
wifi evt: 0
ip:192.168.233.207,mask:255.255.255.0,gw:192.168.233.1
wifi evt: 3
[hostByName] Host: 192.168.233.71 is a IP!
pm open,type:" 0
state: � -> " (!R8)
rm 0
pm close ?
wifi evt: 1
STA disconnect: 6
reconnect
state: " -> 0 (0)
scandone
state: 0 -> " (?8)
state: " -> @ (0)
state: @ -> � (i8)
add 0
aid "
cnt 
state: � -> " (�R8)
rm 0
wifi evt: 1
STA disconnect: 2
reconnect
state: " -> 0 (0)
scandone
state: 0 -> " (?8)
state: " -> @ (0)
state: @ -> � (i8)
add 0
aid "
cnt 
state: � -> " (�R8)
rm 0
wifi evt: 1
STA disconnect: 2
reconnect
state: " -> 0 (0)
scandone
state: 0 -> " (?8)
state: " -> @ (0)
state: @ -> � (i8)
add 0
aid "
cnt 
state: � -> " (�R8)
rm 0
wifi evt: 1
STA disconnect: 2

Endless repeat of reconnect with disconnect: 2 (REASON_AUTH_EXPIRE)

256dpi, 2021-12-06:
Could you make the same test with the arduino-mqtt example sketch?

gskjold, 2021-12-06:
Fixed by removing recently added code for MQTT SSL. Guess it overwrites some memory... https://github.com/gskjold/AmsToMqttBridge/commit/bc6d45ecf2bff127db973dc98f78c7ce0f81bfc1

closed 2021-12-06 by gskjold