256dpi/arduino-mqtt #278: garbage on mqtt string when setting the buffer

Rocco83, 2022-01-28:
Hi,

I have an issue sending out the mqtt message (json).
I believe that i'm simply passing the default buffer size (128 byte if i'm not wrong) summing up the topic & message.
So, I need to enlarge the buffer on the constructor.

The snippet is the following (full code: https://github.com/Rocco83/caldaia):
```
  mqttreturn = mqttClient.publish("/caldaia", "debug");
  if ( !mqttreturn ) { Serial.println(F("MQTT short debug send message failed")); } else { Serial.println(F("MQTT short debug easy string OK")); }
  strcpy(json_string, "{\"temperature\":23.63281,\"temperature_unit\":\"°C\",\"pressure\":-0.014143,\"pressure_unit\":\"°C\",\"mqttRawData\":[131");
  mqttreturn = mqttClient.publish("/caldaia", json_string);
  if ( !mqttreturn ) { Serial.println(F("MQTT long debug send message failed")); } else { Serial.println(F("MQTT long debug easy string OK")); }
```

Without options, this is the footprint on compile:
```
Sketch uses 30952 bytes (95%) of program storage space. Maximum is 32256 bytes.
Global variables use 1156 bytes (56%) of dynamic memory, leaving 892 bytes for local variables. Maximum is 2048 bytes.
```

Here comes the problem:
1) setting the constructor to 160 bytes will produce garbage. Output in mosquitto:
```
debug
{"temperature":23.63281,"temperature_unit":"°C","pressure":-0.014143,"pressure_unit":"°C",*�*�����H
```
This is the footprint on compile:
```
Sketch uses 30952 bytes (95%) of program storage space. Maximum is 32256 bytes.
Global variables use 1156 bytes (56%) of dynamic memory, leaving 892 bytes for local variables. Maximum is 2048 bytes.
```
2) setting the constructor to 256 bytes will not not show anything in mqtt. *No* output in mosquitto (not even the short one, so "debug").
This is the footprint on compile:
```
Sketch uses 30952 bytes (95%) of program storage space. Maximum is 32256 bytes.
Global variables use 1156 bytes (56%) of dynamic memory, leaving 892 bytes for local variables. Maximum is 2048 bytes.
```

As you can see, the footprint is not changing at all, which sounds quite strange to me, as this is a global variable.

I am not sure if this is a bug, but... seems to me so.

Rocco83, 2022-01-31:
Brief notice, increasing the buffer (but lowering the footprint of other variables) fixed the issue.

256dpi, 2023-02-04:
Closing as solved.

closed 2023-02-04 by 256dpi