256dpi/arduino-mqtt #281: Project status

raphael-bmec-co, 2022-04-19:
Hi dev team.

Is this project being actively maintained or is it being shut down? 

Kind regards

256dpi, 2022-04-20:
I guess neither. I do not have the time lately to look at every GitHub Issue and do a problem analysis. Most of the problems are anyway related to the Arduino Networking stack. Since the underlying LWMQTT library is used in many projects it's considered stable. Thus, most of the time there is not much we can do in this project. I'll be looking to finish support for MQTT5, but need to find time to look into that as it touches multiple projects. However, the project is certainly not shut-down. There are already some people replying to GitHub issues and helping others. If somebody has the time to do more community work I'm happy to have them step in anytime.

raphael-bmec-co, 2022-04-20:
Hi @256dpi. I appreciate the response and your position. Many thanks.

closed 2022-04-30 by raphael-bmec-co