256dpi/arduino-mqtt #282: esp32 Serial1 stops working with mqtt

IngLopes, 2022-05-05:
Hey I've got a problem for a few days that is killing me. I want to read distance from a TFmini Lidar sensor with esp32 then send it by mqtt. I can make the readings and my mqtt connection is working, but if i put them both together the sensor doesn't show any readings anymore. Serial1.available() becomes always 0. 
I see no reason why mqtt should interfere with the Serial1 port. Does anyone have any ideas?
 
This is my code only for reading the sensor:
//TEST BASIC READING WITH TFMINI LIDAR
#include <TFMini.h>
#include <HardwareSerial.h>
#include <PubSubClient.h>
#include <WiFi.h>
// Note the format for setting a serial port is as follows: Serial2.begin(baud-rate, protocol, RX pin, TX pin);
#define RXD2 18
#define TXD2 19
#define TFMINI_BAUDRATE 115200
TFMini tfmini;
uint16_t dist;
uint16_t strength;
String s_dist, s_strength;
//Para conectar a rede movel celular
const char* ssid = "Ing_LM";
const char* password = "eqtc1876";
const char* host = "192.168.43.161"; // Servidor (Apache)
//Check if bluetooth configurations are enabled
//#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
//#error Bluetooth is not enabled! Please run 'make menuconfig' to and enable it
//#endif

//BluetoothSerial SerialBT;
WiFiClient TCP_Client;
PubSubClient client(TCP_Client);
void setup() {
  
  Serial.begin(TFMINI_BAUDRATE,SERIAL_8N1);
//  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("Bluetooth started, ready to pair...");
  Serial.println("\nBenewake TFmini UART LiDAR Program");
  Serial.println ("Initializing...");
  Serial1.begin(TFMINI_BAUDRATE, SERIAL_8N1, RXD2, TXD2);
  //Serial2.begin(TFMINI_BAUDRATE,SERIAL_8N1);   
  
  // Initialize the TF Mini sensor
  tfmini.begin(&Serial1);    

   //Initializing wifi and mqtt server
  WiFi.mode(WIFI_STA);
  //Stabilishing wifi connection:
  Serial.print("Conectando: ");
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("Ligado a wifi");
  client.setServer(host,1883);
  client.connect("esp32");
  Serial.println("esp connected to server");
  
}

void loop() {
  if(Serial1.available()>1){  // GPS
  dist = tfmini.getDistance();
  strength = tfmini.getRecentSignalStrength();
  //Send measurements by bluetooth
  //SerialBT.print(dist);
  //SerialBT.print(", ");
  //SerialBT.println(strength);
  
 // Display the measurement
  Serial.print(dist);
  Serial.print(" cm      sigstr: ");
  Serial.println(strength);
  Serial1.flush();

  }

  // Wait some short time before taking the next measurement
  //delay(25);  
}

And this is my whole code: with the connection to HiveMQ Mqtt cloud broker
//TESTING TO CONNECT ESP32 AS A CLIENT TO CLOUD MQTT BROKER USING HIVEMQCLOUD

//INCLUDE NECESSARY LIBRARIES
#include <WiFi.h>
#include <PubSubClient.h>
#include <WiFiClientSecure.h>
#include <TFMini.h>
#include <HardwareSerial.h>

//FIRST STEP - connect to the wifi
//--Wifi settings
const char* ssid = "Ing_LM";
const char* password = "eqtc1876";

//---Function for connecting to wifi
void connect_wifi(){
  WiFi.mode(WIFI_STA);
  //Stabilishing wifi connection:
  Serial.print("Conectando: ");
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  // O ESP envia por porta série, para o PC local esta informaçao:
  Serial.println("");
  Serial.println("O ESP está ligado por WiFi ao Router, o Router deu-lhe o IP");

  // IP address
  Serial.print("IP local: ");
  Serial.println(WiFi.localIP()); // Imprimir IP do ESP
}

//SECOND STEP - Configure client esp32 mqtt

WiFiClientSecure wifiClient;
PubSubClient client(wifiClient); 

//MQTT Cloud Broker settings
const char *mqtt_server = "6e42497930bd4bd99a0aa8ac31773d8f.s1.eu.hivemq.cloud";
const int mqtt_port = 8883;
const char *mqtt_user = "LidarS";
const char *mqtt_pass = "****************";
const char *mqtt_client_name = "Esp32Lidar"; // Client connections cant have the same connection name

// THIRD STEP - Configure the connection with the TFmini Lidar Sensor
#define RXD2 16
#define TXD2 17
#define TFMINI_BAUDRATE 115200
TFMini tfmini;
//Measurement variables
int dist, strength;
String s_dist, s_strength;

//CA Certificate

static const char *root_ca PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIFazCCA1OgAwIBAgIRAIIQz7DSQONZRGPgu2OCiwAwDQYJKoZIhvcNAQELBQAw
TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh
cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTUwNjA0MTEwNDM4
WhcNMzUwNjA0MTEwNDM4WjBPMQswCQYDVQQGEwJVUzEpMCcGA1UEChMgSW50ZXJu
ZXQgU2VjdXJpdHkgUmVzZWFyY2ggR3JvdXAxFTATBgNVBAMTDElTUkcgUm9vdCBY
MTCCAiIwDQYJKoZIhvcNAQEBBQADggIPADCCAgoCggIBAK3oJHP0FDfzm54rVygc
h77ct984kIxuPOZXoHj3dcKi/vVqbvYATyjb3miGbESTtrFj/RQSa78f0uoxmyF+
0TM8ukj13Xnfs7j/EvEhmkvBioZxaUpmZmyPfjxwv60pIgbz5MDmgK7iS4+3mX6U
A5/TR5d8mUgjU+g4rk8Kb4Mu0UlXjIB0ttov0DiNewNwIRt18jA8+o+u3dpjq+sW
T8KOEUt+zwvo/7V3LvSye0rgTBIlDHCNAymg4VMk7BPZ7hm/ELNKjD+Jo2FR3qyH
B5T0Y3HsLuJvW5iB4YlcNHlsdu87kGJ55tukmi8mxdAQ4Q7e2RCOFvu396j3x+UC
B5iPNgiV5+I3lg02dZ77DnKxHZu8A/lJBdiB3QW0KtZB6awBdpUKD9jf1b0SHzUv
KBds0pjBqAlkd25HN7rOrFleaJ1/ctaJxQZBKT5ZPt0m9STJEadao0xAH0ahmbWn
OlFuhjuefXKnEgV4We0+UXgVCwOPjdAvBbI+e0ocS3MFEvzG6uBQE3xDk3SzynTn
jh8BCNAw1FtxNrQHusEwMFxIt4I7mKZ9YIqioymCzLq9gwQbooMDQaHWBfEbwrbw
qHyGO0aoSCqI3Haadr8faqU9GY/rOPNk3sgrDQoo//fb4hVC1CLQJ13hef4Y53CI
rU7m2Ys6xt0nUW7/vGT1M0NPAgMBAAGjQjBAMA4GA1UdDwEB/wQEAwIBBjAPBgNV
HRMBAf8EBTADAQH/MB0GA1UdDgQWBBR5tFnme7bl5AFzgAiIyBpY9umbbjANBgkq
hkiG9w0BAQsFAAOCAgEAVR9YqbyyqFDQDLHYGmkgJykIrGF1XIpu+ILlaS/V9lZL
ubhzEFnTIZd+50xx+7LSYK05qAvqFyFWhfFQDlnrzuBZ6brJFe+GnY+EgPbk6ZGQ
3BebYhtF8GaV0nxvwuo77x/Py9auJ/GpsMiu/X1+mvoiBOv/2X/qkSsisRcOj/KK
NFtY2PwByVS5uCbMiogziUwthDyC3+6WVwW6LLv3xLfHTjuCvjHIInNzktHCgKQ5
ORAzI4JMPJ+GslWYHb4phowim57iaztXOoJwTdwJx4nLCgdNbOhdjsnvzqvHu7Ur
TkXWStAmzOVyyghqpZXjFaH3pO3JLF+l+/+sKAIuvtd7u+Nxe5AW0wdeRlN8NwdC
jNPElpzVmbUq4JUagEiuTDkHzsxHpFKVK7q4+63SM1N95R1NbdWhscdCb+ZAJzVc
oyi3B43njTOQ5yOf+1CceWxG1bQVs5ZufpsMljq4Ui0/1lvh+wjChP4kqKOJ2qxq
4RgqsahDYVvTH9w7jXbyLeiNdd8XM2w9U/t7y0Ff/9yi0GE44Za4rF2LN9d11TPA
mRGunUHBcnWEvgJBQl9nJEiU0Zsnvgc/ubhPgXRR4Xq37Z0j4r7g1SgEEzwxA57d
emyPxgcYxn/eR44/KJ4EBs+lVDR3veyJm+kXQ99b21/+jh5Xos1AnX5iItreGCc=
-----END CERTIFICATE-----
)EOF";

void reconnectMQTT() {
  //Loop until reconnected
  while (!client.connected()){
    Serial.print("Attempting connetion to MQTT Broker...");
    //Attempt to connect
    if (client.connect(mqtt_client_name, mqtt_user, mqtt_pass)){
      Serial.println("Connected.");
      client.subscribe("Distance"); //subscribe to topic Distance
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.print(" try again in 2 seconds");
      delay(2000);
    }
  }
}

void setup(){
  Serial.begin(TFMINI_BAUDRATE);
  Serial1.begin(TFMINI_BAUDRATE, SERIAL_8N1, RXD2, TXD2);
  // Initialize the TF Mini sensor on Serial1
  tfmini.begin(&Serial1);  
  // Initializing wifi and mqtt server
  connect_wifi();
  wifiClient.setCACert(root_ca); //for secure connection
  client.setServer(mqtt_server, mqtt_port);
  client.setCallback(callback);  
}

void loop(){
  if (!client.connected()) reconnectMQTT();
  client.loop();
  
  if (Serial1.available()>1){
    dist=tfmini.getDistance();
    s_dist = String(dist);
    Serial.print("Publish dist on MQTT: ");
    Serial.println(dist);
    client.publish("Distance", s_dist.c_str());
    
  } else {
    Serial.println(Serial1.available());
  }
}

//The void callback is called every time we have a message from the broker
void callback(char* topic, byte* payload, unsigned int length) {
  String incommingMessage = "";
  for (int i = 0; i < length; i++) incommingMessage+=(char)payload[i];
  
  Serial.println("Message arrived ["+String(topic)+"]"+incommingMessage);
  }

   //  check for other commands
  }

IngLopes, 2022-05-05:
I tried taking out #include <HardwareSerial.h> cause its not doing anything but that wasnt the problem either

256dpi, 2023-02-04:
Closing as probably not related to this library.

closed 2023-02-04 by 256dpi