256dpi/arduino-mqtt #291: Unable to connect with Platform I/O espressif8266@4.1.0 (Arduino Core v3.1)

Johboh, 2023-01-21:
Hi!
I have not investigated this further and will provide more debug printouts, but everything works fine when using version [2.5.0](https://github.com/256dpi/arduino-mqtt/releases/tag/v2.5.0) of this library and [espressif8266@4.0.1](https://github.com/platformio/platform-espressif8266/releases/tag/v4.0.1) which uses [Arduio Core 3.0.2](https://github.com/esp8266/Arduino/releases/tag/3.0.2). 
But after upgrading to [espressif8266@4.1.0](https://github.com/platformio/platform-espressif8266/releases/tag/v4.1.0) which uses [Arduino Core 3.1.1](https://github.com/esp8266/Arduino/releases/tag/3.1.1), I'm unable to connect to the MQTT server with the error message `LWMQTT_MISSING_OR_WRONG_PACKET = -9`.

This is mostly a heads-up that if anyone else is experience connection issues when using Platform I/O and [espressif8266@4.1.0](https://github.com/platformio/platform-espressif8266/releases/tag/v4.1.0), try with [espressif8266@4.0.1](https://github.com/platformio/platform-espressif8266/releases/tag/v4.0.1) until this issue has been solved (if the issue is in this library or in Arduino Core, we don't know yet).

mp-se, 2023-01-26:
Same as #292

mp-se, 2023-01-29:
There is a solution in #292. Changing delay(0) to delay(1) fixes the issue with latest arduino core.

256dpi, 2023-02-04:
Closing in favour of #292.

closed 2023-02-04 by 256dpi