256dpi/arduino-mqtt #295: Can't get connect with AWS IoT Core

amid0, 2023-02-09:
Trying to connect NodeMCU esp8266 board to AWS IoT core MQTT broker with certificates.
And its not working.
So the problem that it is newer gets connect method return true.
`
void setup()
{
  NTPConnect();
  client.setTrustAnchors(&CACertificate);
  client.setClientRSACert(&ClientCertificate, &ClientPrivateKey);
  
  clientMQTT.begin(MQTT_HOST, 8883, client);
  Serial.println("Connecting to AWS IOT");

  while (!clientMQTT.connect(THINGNAME)) {
    Serial.printf("clientMQTT.lastError = %d\n", clientMQTT.lastError());
    Serial.printf("clientMQTT.returnCode = %d\n", clientMQTT.returnCode());
    delay(30000);
  }

  if(!clientMQTT.connected()){
    Serial.println("AWS IoT Timeout!");
  }
}
`
here is output example 
`
Setting time using SNTP.done!
Current time: Thu Feb  9 18:08:28 2023
Connecting to AWS IOT
BSSL:_connectSSL: start connection
BSSL:Connected!
BSSL:read: Not connected, none left available
clientMQTT.lastError = -5
clientMQTT.returnCode = 6
`
clientMQTT.lastError = -5 - LWMQTT_NETWORK_FAILED_READ
clientMQTT.returnCode = 6 - LWMQTT_UNKNOWN_RETURN_CODE

Any suggestions?
Spend two days to digging into it and run out of ideas  (
Seems like it sends some data but newer read

256dpi, 2023-02-10:
Make sure you use the latest version of this library "2.5.1". This may be related to #292.

amid0, 2023-02-10:
Seems like found out the reason - its AWS IAM policies.
Not all required permissions was granted and its lead to situation when nothing is work and no meaningful errors  ) 
I'm not following pattern that is described in all guides - just assign admin permissions and forget )

closed 2023-02-10 by amid0