256dpi/arduino-mqtt #300: MQTT reconnection when not disconnected. ESP8266

gleanlux, 2023-03-16:
Hello!

I have a problem that I have been trying to solve for a long time.

An ESP feather board (ESP8266) reconnects to MQTT at completely random intervals.
Looking at the mosquito log, there is no disconnect, only reconnect, so when reconnecting there is an immediate disconnect/connect from the server.

In the code, as recommended, there is a 10ms delay in the loop after the MQTT loop, no delay in the rest of the program.

I have placed several debug elements in the code to detect the error.
- No interruption on the wireless station side.
- ESP wireless side no connection failure.

According to the latest debug, I retrieved the lastError() parameter on MQTT disconnection, which returns the following error each time and until reconnection:
"LWMQTT_MISSING_OR_WRONG_PACKET = -9"

Configuration:
setKeepAlive(15);
or the timeout was increased last time:
setTimeout(2000);

Is there anything I haven't tried yet that might help solve the error?

Thank you very much!

crazy888, 2024-01-29:
I have the same problem. 
!client.connect is false but my mosquitto logs says there is no disconnect of that client.
So the ESP reconnect a lot without any reason.

`1706501287: New connection from 192.168.1.192:50896 on port 1883.
1706501287: Client Piridach already connected, closing old connection.
`

I also tried the set keepalive up and also moved my mosquitto server from my pi to my nas, but it does not change anything.
I have other clients like tasmota connected to the mosquitto server without any problems.
Also my network is stable without any drops.
It has nothing to do with publish things. It just happens in standby while looping over client.loop(); 
So i´m out of ideas...........

Afriza87, 2024-05-19:
Make sure the client IDs are not the same, please.

When using "Arduino," it's better to use a client ID that no one else is likely to use.

If there is a duplicate client ID, it will immediately disconnect from the broker, and because it disconnects, the ESP32 will attempt to reconnect.

This will continue endlessly. Additionally, if it is used for subscribing, just subscribe to one topic using a wildcard, for example, ex/#. This way, all topics starting with ex will automatically be subscribed.

gleanlux, 2024-05-21:
In my case the problem was caused by QOS1. 
Currently, the library cannot handle QOS1, and this is what caused me to lose the connection. 
Since I set it to QOS0 the problem has disappeared.

raphael-bmec-co, 2024-07-02:
@gleanlux I am observing the same. Is it documented that QOS1 is not supported?

raphael-bmec-co, 2024-07-02:
Actually on closer inspection it seems like making it QOS0 just increases the likelihood that the publish will succeed. If it fails the connection is closed and that is the root cause. It would be good to understand why this [line](https://github.com/256dpi/arduino-mqtt/blob/a1ae6d343074f882bc2c9a39d0dc7f680a4c4e50/src/MQTTClient.cpp#L408) is here.