256dpi/arduino-mqtt #307: This library uses some posix functions that are not provided for Uno R4

nelseric, 2023-07-07:
See: https://github.com/arduino/ArduinoCore-renesas/issues/18
When I try to compile this library for my Uno R4, I get the following errors:
```
C:\Users\Eric Nelson\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp: In member function 'void MQTTClient::setHost(const char*, int)':
C:\Users\Eric Nelson\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp:259:20: error: 'strdup' was not declared in this scope
   this->hostname = strdup(_hostname);
                    ^~~~~~
C:\Users\Eric Nelson\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp:259:20: note: suggested alternative: 'strcmp'
   this->hostname = strdup(_hostname);
                    ^~~~~~
                    strcmp
C:\Users\Eric Nelson\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp: In member function 'void MQTTClient::setWill(const char*, const char*, bool, int)':
C:\Users\Eric Nelson\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp:277:37: error: 'strdup' was not declared in this scope
   this->will->topic = lwmqtt_string(strdup(topic));
                                     ^~~~~~
C:\Users\Eric Nelson\Documents\Arduino\libraries\MQTT\src\MQTTClient.cpp:277:37: note: suggested alternative: 'strcmp'
   this->will->topic = lwmqtt_string(strdup(topic));
                                     ^~~~~~
                                     strcmp
```

Hsubtnarg, 2023-07-21:
I have the same issue with my new Uno R4. This library is great, much easier to use than the others, but not on a R4

256dpi, 2023-08-09:
If https://github.com/arduino/ArduinoCore-renesas/pull/87 is merged, the problem should go away.

256dpi, 2024-02-10:
The mentioned PR has been merged and I just compiled this library with the latest core for the UNO R4 and it seems to work. But I haven't tested with a real board tough. Please re-open if there are still any issues.

closed 2024-02-10 by 256dpi