256dpi/arduino-mqtt #314: RP2040 seems to work fine

savejeff, 2024-01-13:
Just wanted to inform that i tested arduino-mqtt on RP2040 / Raspberry Pi Pico W and with the earlephilhower/arduino-pico Core [Here](https://github.com/earlephilhower/arduino-pico) it works out of the box no issues!

Would be cool to have this added to the Readme, as it currently is not officially stated to work with RP2040

Great stuff

256dpi, 2024-02-10:
Would you be up for contributing an example for the RP2024?

savejeff, 2024-02-10:
the examples run out of the box, no midifications needed. You have to use the earlephilhower/arduino-pico and just compile. 

For Platformio here an example configuration: 

[env:BOARD_RP2040_PICO_W]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages = framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
board = rpipicow
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m