256dpi/arduino-mqtt #316: Random error in  MQTTClient::loop() function

martinch14, 2024-02-08:
´´´
  // get available bytes on the network
  auto available = (size_t)this->netClient->available();

  // yield if data is available
  if (available > 0) {
    this->_lastError = lwmqtt_yield(&this->client, available, this->timeout);
    if (this->_lastError != LWMQTT_SUCCESS) {
      // close connection
      this->close();

      return false;
    }
  }
´´´
I am testing an mqtt broker, and I have random disconnection problems, I am sending one message per second, sometimes the disconnection happens quickly, sometimes I send up to 1000 messages without problems.

The disconnection event occurs when the call to the function (size_t)this->netClient->available();

Results in the value
available:4294967220

This causes it to enter the loop and then close the connection,

If I comment this line of code, I can send without problems and without losing messages,

Do you have any idea what could be happening?

I can detect the disconnection and reconnect, but I would like to avoid it, since it is the device that is disconnecting, it is not a question of internet access either, because I try it with different devices and the disconnections occur at different times, It is the device that disconnects from the broker

´´´

256dpi, 2024-02-10:
Thanks for the report, the included snippet made me think and I found an integer casting issue:

```
auto available = (size_t)this->netClient->available();
```

The `available` method actually returns an `int`, so my feeling is that your network layer may return a negative number to indicate an error code. When casting it to `size_t` the number becomes very large due to "Two's Complement".

I fixed the issue and released a new version. Let me know if it helps.

martinch14, 2024-02-19:
Hi @256dpi 
Apply the mentioned changes and uncomment that part of the code in my firmware.
I've been testing this for a few days and for now I don't have any disconnection events, I think the problem is solved.

Thank you so much

Decezaris, 2024-05-27:
@256dpi, shoudn't lwmqtt_yield and its nested functions change the type of `available` parameter to int as well?

closed 2024-08-30 by 256dpi